cta-observatory/ctapipe #1506: Remove some tools

maxnoe, 2020-11-05:
I think we have a lot of tools that are not actually needed and also don't really follow current best practices.

Easiest would be just to remove them:

- [ ] display_events_single_tel
- [ ] display_integrator
- [ ] display_summed_images
- [ ] camdemo

kosack, 2020-11-06:
We can probably remove some of those, even if some are useful for debugging.   I use display_summed_images and display_events_single_tel frequently for making animations for presentations,  but they could be placed in my own private repo so no problem removing them unless others use them.  I'd keep camdemo, as it was meant to be a general test of how the config, Tools,  and graphics works, so that is harmless and nice to have to demonstrate that things are working (it's the first think I run on a new install or to show people what data looks like).

maxnoe, 2020-11-06:
Aren't these usecases also supported by `display-dl1` and the `bokeh` viewers?

kosack, 2020-11-09:
> Aren't these usecases also supported by display-dl1 and the bokeh viewers?

not completely, but would be pretty easy to merge all into one tool

maxnoe, 2020-11-09:
I am in the process of rewriting the bokeh viewer. Can you make a list of features we want? 
Then I will remove the old tools when their features are included in the bokeh viewer.

I also want to make the bokeh viewer a component that can be optionally enabled in e.g. the stage1 tool to debug things.

moralejo, 2020-11-10:
> I am in the process of rewriting the bokeh viewer. Can you make a list of features we want?
> Then I will remove the old tools when their features are included in the bokeh viewer.
>
> I also want to [...]

Hi, a couple from my side, for the camera display, that we implemented in [https://github.com/cta-observatory/cta-lstchain/blob/master/lstchain/visualization/bokeh.py](url):

- lin/log version of a display, using tabs
- Slider to modify lower and upper end of the z-scale
- accompanying histogram of all camera values and (perhaps) plot of camera values vs. pixel id
- Pixel info shown (id, content, whatever - configurable) when mouse tip hovers over it
- Optional slider to show different versions of the  displayed data. We use it for example to show different runs (when displaying e.g. mean pixel charge in a run)

maxnoe, 2023-10-21:
done

closed 2023-10-21 by maxnoe