cta-observatory/ctapipe #1508: SimTelEventSource should not record true_image if it isn't in the simulation

kosack, 2020-11-06:
SimTelEventSource writes an array of 0s to `event.simulation.tel[x].true_image`  (photoelectrons with no noise) in the case that no true_image exists in the data .  I think it would be better behavior to just set true_image to None.  Was there a reason for this?  If not, we have to check that `true_image.sum()>0` in order to determine if the file contains true_images, which is slow and a bit confusing. 

This lead to a bug in DL1EventSource not being found:
In this line
https://github.com/cta-observatory/ctapipe/blob/337a8f056bc6276316d22925dd2242f895b48460/ctapipe/io/dl1eventsource.py#L336

the true image is read as "image" instead of "true_image", which is a bug that somehow never gets tested properly (I think because the test file has no true_images).

maxnoe, 2020-11-06:
The reason was that pyhessio did this and we strived for compatibility.

maxnoe, 2020-11-06:
I agree, that we should not do this anymore, and there are probably other fields where this is done.

kosack, 2020-11-06:
I was actually a bit confused as to how the `test_dl1eventsource.test_simulation_info` test was passing... When I run ctapipe-stage1 on *gamma_test_large.simtel.gz*, I get all 0s for the true_image (since it doesn't exist in the input), but then the test looks for any value being non-zero, which shouldn't be the case. It seems to fail only if we write out the array with all 0s to the file, but if nothing is written (no simulation/event/telescope dataset), it passes... somehow calling np.any on a value of None works or something like that.

maxnoe, 2023-01-04:
This was fixed some time ago, true_image now is None if the data is not present in the input file.

closed 2023-01-04 by maxnoe