cta-observatory/ctapipe #1510: Participate in iminuit v2.0 beta?

HDembinski, 2020-11-06:
Dear ctapipe team,

I am about to finish a major rewrite of iminuit, version 2.0, that replaces Cython as the tool to wrap C++ Minuit2 with pybind11, which is going to solve several issues that the legacy code had. All the good things that this will bring are listed on top of this PR:
https://github.com/scikit-hep/iminuit/pull/502

Switching to the new version of iminuit should be completely transparent to you, since the new version passes the comprehensive suite of unit tests of iminuit-v1.x. However, I would like to use this opportunity to finally remove interface that has been successively marked as deprecated in versions 1.3 to 1.5.

Therefore my two question to you:
- Did you take note of the deprecation warnings in iminuit and did you keep up with the interface changes?
- Are interested in trying out a Beta release of v2.0 to work out any possible bugs in the new version before the release?

Best regards,
Hans, iminuit maintainer

kosack, 2020-11-09:
Great! I don't think we use any deprecated features, but it should be easy to check with your PR.   

For the ctapipe devs: So far `iminuit` is only used in two places: *muon ring fitting*, and in *ImPACT reconstruction*.  It's pretty easy to test muon ring fitting, but not the ImPACT reconstruction, which lacks some unit tests and data. @ParsonsRD do you foresee any problems updating to iminuit 2?

maxnoe, 2020-11-09:
I would propose to remove the impact code and re-add it together with unit tests / updated API when someone wants to put in the work for that.

Right now, it is basically unmaintained and untested without a clear path how even create the templates / add tests.

ParsonsRD, 2020-11-10:
@kosack Minuit is only an option for the ImPACT code so can be easily checked or removed if this breaks any functionality, it works just fine with the scipy optimisers.

HDembinski, 2020-11-12:
Note that you can do `from iminuit import minimize` which mimics scipy's minimize. If you use that, you have zero adaptation issues.

HDembinski, 2020-12-08:
iminuit v2.0 is released, the breaking changes in the interface are documented in the [changelog](https://iminuit.readthedocs.io/en/stable/changelog.html) which can serve as an upgrade guide.

PR #1714 (2021-05-11): Use iminuit>=2
Closes #1510
closed 2021-05-11 by maxnoe