cta-observatory/ctapipe #1518: how to apply parameter quality cuts in stage2

kosack, 2020-11-16:
The sequence envisioned for the stage2 processing is:

```py3
for event in source:
     calibrate(event) # only if not already dl1
     process_images(event)  # cleans images, applies Image QualityQuery, and parmeterizes images
     # at this point, non-passing parameters are NaN

     # the part we want to implement now:
     # ------------------------------------
     apply_parameter_quality_cuts(event) # e.g. ellipticity, leakage. Where do these results go?
     reconstruct(event)

     write_event(event)
```

##  Problem
between `apply_parameter_quality_cuts` and `reconstruct`, there needs to be a place in the `ArrayEvent` data structure to tell reconstruct() which images to use. 

* We don't really want to drop `event.dl1.tel.parameters` as they then wouldn't get written in the case we output both DL1 and DL2 data, so just using `event.dl1.tel.values()`  is not enough
* if more than one reconstructor is used, it could be that we need different cuts for each (ImPact for example can use truncated images)
* checking for `parameters.hillas.intensity=NaN` is fine to see if the parameters exist, but not that they passed quality cuts

## Possible Solution 1:
* add a field to `ImageParametersContainer` for parameter quality flags.  
* use the Parameters `QualityQuery` to write this (it outputs a boolean array of which cuts passed), so you not only get a "good/bad" flag, but in fact the flag for each cut individually. This looks a lot like the `event.trigger.tels_with_trigger` flags, but for a single telescope and each bit means a cut passing or not. 
* in HillasReconstructor, you can then just look at this quality flags field and include only telescopes where it passes fully, e.g.:
```py3
hillas_dict = {tid,p.hillas for tid, p in event.dl1.tel.items() if np.all(p.quality==True)}
```
(assuming we define True meaning pass)

## Possible solution 2:

This is still a bit difficult if there are more than one reconstructor (e.g. Hillas + ImPACT)  that want different cuts, so another option would be to simply include the QualityQuery that operates on images *inside* each Reconstructor (so it is responsible for determining which parameters to use, like the ImageProcessor with images).  I like that, but then it would still be nice to write out the event-wise quality flags, so not sure where those would go.

kosack, 2020-11-16:
I think in the end this issue is really two things:

1. where do we apply cuts on parameters (inside or outside the Reconstructor)
2. should we also record the event-wise output of the quality queries in the DL1 or DL2 files (now we just record statistics for DL1)?  If so where should they go in the event structure and output tables?

maxnoe, 2020-11-16:
> where do we apply filters on telescope-wise parameters (inside or outside the Reconstructor)

I would say inside makes more sense, since then the different extractors could have different queries,  the users of these methods might consider different cuts appropriate for different algorithms.

kosack, 2020-11-16:
Yes, I agree - it's then much easier to have different cuts for each reconstructor.  

Then question 2 still stands: do we want to store event-wise quality flags, or only statistics as we have now? That would be useful for debugging and benchmarking, but perhaps is not so important (can always be done in a debugging log).

HealthyPear, 2020-11-16:
>     2\. should we also record the event-wise output of the quality queries in the DL1 or DL2 files (now we just record statistics for DL1)?  If so where should they go in the event structure and output tables? Those would be columns of boolean flags (1 flag per cut)

I don't think the values per-se need to be recorded (aren't they the image parameters already?)
I would do something like in protopipe, where e.g. I have 1 column called `good_image` which tells me if an image was used by the reconstructor

we could have a columns per flag and a condition for which if all flags are true then the event is recostructed

kosack, 2020-11-16:
That's basically the same issue: either it's a boolean column per reconstructor (`True`=image was used), or it's a boolean array column of each cut `[True,True,False]`=3rd cut failed, image not used.  Still need to decide where to put them, as these are per-telescope quantities, so should be in `dl2.tel[x]` I think.

HealthyPear, 2020-11-16:
I guess one column with a boolean array wouldn't be a bad choice

if it will be linked to the reconstruction algorithm then yes it needs to stay in dl2 I guess

what about `/dl2/event/telescope/tel_XXX/<RECO_ALGO_NAME>/cutname` ?

kosack, 2020-11-16:
See above, I think we should be more general, as we want to include also mono DL2 parameters computed from the DL1 information, optionally.  So it would go there. I think having the algo name as the prefix is probably best as well, rather than making too many tables, but we can see

maxnoe, 2022-12-08:
We decided to go for the approach that each reconstructor has its own quality query in recent releases, closing for now.

closed 2022-12-08 by maxnoe