cta-observatory/ctapipe #1528: Tools do not log traceback on exceptions

maxnoe, 2020-11-17:
For some reason, although `Tool` uses

```
            self.log.exception(f"Caught unexpected exception: {err}")
```

No traceback is logged when an exception  occurs.

kosack, 2020-11-18:
might need to add `exc_info=True` to the call

maxnoe, 2020-11-18:
No, does not work. Also `log.exception` has `exc_info=True`, it basically is `log.error` with `exc_info=True`

kosack, 2020-11-18:
I think the traceback output is handled by the logging Formatter - perhaps the custom formatter we use is missing this.

kosack, 2020-11-18:
maybe it just need to be added to the default format string for the logger.  I notice that LogRecords contain an sinfo dict, but that is not in our log format:

https://docs.python.org/3/library/logging.html#logrecord-objects

maxnoe, 2020-11-20:
The formatter implemented a custom formatting instead of using the one of the base class and just adding the color information.

commit f224a08b (2020-11-23): Fix missing traceback in colored formatter, fixes #1528
commit ffc06f7c (2020-11-23): Fix missing traceback in colored formatter, fixes #1528 (#1541)

* Fix missing traceback in colored formatter, fixes #1528

* Make sure record is not mutated

* Update logging.py
closed 2020-11-23 by maxnoe