cta-observatory/ctapipe #1532: parallel=True not efficient or faster (in some cases)

kosack, 2020-11-19:
This is just to summarize an offline conversation: 
## Problem: 

Doing some benchmarking I found running `ctapipe-stage1` on a large number of events using 2 cases on a machine with a 6-core (12 with hyperthreading) Intel core i7 coffee lake processor:

1. standard case (just running it as usual)
2. setting `NUMBA_NUM_THREADS=1`

* Both analyses finish at the same time (*speed is essentially identical*)
* in case 1, all 6 processors are at 100% the whole time (and the fan kicks in quickly)
* in case 2, only one processor is used, and is at around 60-90% (rest is probably IO), no fan

This means that the parallelism used in the numba calls in CameraCalibrator does not help the speed, but rather decreases the efficiency (lots more CPU power used with the same result).   

This is likely true because the size of the arrays being looped over (N_pixels * N_samples * N_channels) is actually not very large from a parallelization standpoint, so the act of distributing data to multiple threads has more overhead than the speed increase provided.   It may not be the case on slower processors or those without good vector processing instructions. 

## Options:
- leave things as they are, but recommend running `NUMBA_NUM_THREADS=1 ctapipe-stage1 ...`, so one can optimize to each processor
- just remove the `parallel=True` from the `@njit()` decorators so the default is to run in single core mode for these functions.

kosack, 2020-11-19:
note this may also depend on which parallelization backend is used  and the options used, so merits some more study (see https://numba.pydata.org/numba-doc/dev/user/threading-layer.html).

watsonjj, 2020-11-19:
Isn't there another explanation, that the CameraCalibrator is not the primary bottleneck in `ctapipe-stage1`, irregardless of the `parallel=True`? And just for reference, `parallel=True` is only used for one method in ctapipe, `neighbor_average_waveform`. One would need to benchmark this method directly to observe the advantage given. If I remember correctly, it is an improvement, but probably not to a level that a bottleneck is created without it.

kosack, 2020-11-20:
It's true it is not the main bottleneck anymore - in fact parameterization is.  But that is the only place I see parallel=True being used.  The problem is not the overall speed, just the CPU efficiency (seems we are just wasting cycles doing parallelization).  However, I'm still not totally convinced that it needs to be removed - I still have to do a bit of testing to see if we just need to add TBB or OMP as a dependency to get it to  be more efficient.

maxnoe, 2022-04-13:
I tested this again just now and it's still the same. Running with many threads completely saturates the machine without any speedup. We should just remove it.

PR #1878 (2022-04-13): Remove parallel=True for neighbor_waveform_sum, fixes #1532
fixes #1532
closed 2022-04-13 by kosack