cta-observatory/ctapipe #1534: ctapipe-merge examples not working

bregeon, 2020-11-20:
```
(ctapipe_v0.10.0) bash-4.2$ ctapipe-merge -h
Merges DL1-files from the stage1-process tool
(...)

   If you want a specific file pattern as input files, use --pattern='pattern.*.dl1.h5',
    e.g.:
    
    > ctapipe-merge --output=/path/output_file.h5 --progress --pattern='pattern.*.dl1.h5'
```
but:
```
(ctapipe_v0.10.0) bash-4.2$ ctapipe-merge --output=Data/merged_DL1.h5 --progess --pattern='gamma_20deg_0deg_run18*___cta-prod5-lapalma_desert-2158m-LaPalma-dark.h5'
usage: ctapipe-merge [-h] [--skip-images] [--skip-simu-images]
                     [--skip-parameters] [--skip-broken-files] [--overwrite]
                     [--progress] [-q] [--no-q]
                     [--input-dir MergeTool.input_dir]
                     [-i MergeTool.input_dir] [--output MergeTool.output_path]
                     [-o MergeTool.output_path]
                     [--pattern MergeTool.file_pattern]
                     [-p MergeTool.file_pattern]
                     [--config MergeTool.config_file]
                     [--log-level MergeTool.log_level]
                     [--log-file MergeTool.log_file]
                     [--log MergeTool.log_file] [-l MergeTool.log_file]
                     [--log-file-level MergeTool.log_file_level]
                     [extra_args [extra_args ...]]
ctapipe-merge: error: argument --progess: expected one argument

```

bregeon, 2020-11-20:
```
(ctapipe_v0.10.0) bash-4.2$ ctapipe-merge --output=Data/merged_DL1.h5 --pattern='gamma_20deg_0deg_run18*___cta-prod5-lapalma_desert-2158m-LaPalma-dark.h5' --progress
2020-11-20 10:28:26,125 CRITICAL [ctapipe.ctapipe-merge] (dl1_merge.setup): No input files provided, either provide --input-dir or input files as positional arguments

```

maxnoe, 2020-11-20:
The second error is pretty clear, right? If you want to use `--pattern` you also need to provide `--input-dir`

bregeon, 2020-11-20:
```
(ctapipe_v0.10.0) bash-4.2$ ctapipe-merge --output=Data/merged_DL1.h5 --input-dir='./' --pattern='gamma_20deg_0deg_run18*___cta-prod5-lapalma_desert-2158m-LaPalma-dark.h5' --progress
Merging:   0%|                                                                                 | 0/3 [00:00<?, ?Files/s]

```
so `--progress` needs to be the last argumen
and `--pattern` needs `--input-dir`

bregeon, 2020-11-20:
it's just a matter of fixing the doc string, examples given should work :)

maxnoe, 2020-11-20:
Ah, right, lying docs, sorry didn't catch that.

I am confused about `--progress`, order should not be important.

kosack, 2020-11-20:
> --pattern='pattern.*.dl1.h5'

I think you mean:

`--pattern='*.dl1.h5'` (otherwise it won't match anything)

kosack, 2020-11-20:
For me this works:

```sh
ctapipe-merge --output=output_file.h5 --progress --pattern='*.dl1.h5' --input-dir ~/Data/CTA/Prod3b/LaPalmaBaselineNSBscan
```

PR #1551 (2020-11-24): Update MergerTool examples, adding allowed_tels and tests. Fixes #1533 and #1534
- I've updated the examples to fix #1534 and added an error if no output path is given to fix #1533.
 - There is now a new traitlet for `allowed_tels`
 - I also updated the test for the input_dir/pattern/(and now)allowed_tels case. I've seen that @LukasNickel already changed the tests in #1546. I'm not really sure what's now the best way to do it with github - just wait til one branch is merged and then adapt the other? Or should i merge the other one into this one? (For now i just adapt the test_merge so that it  [... cut]
closed 2020-11-24 by maxnoe