cta-observatory/ctapipe #1535: stage1 performance investigation

maxnoe, 2020-11-20:
I profiled the stage1 tool of 0.10.0 to learn where we should look for possible optimizations next.

I used a gamma file from the Prod3b La Palma benchmark. Here is a zipfile with the profile:
[stage1.prof.zip](https://github.com/cta-observatory/ctapipe/files/5572376/stage1.prof.zip)

First couple of rows sorted by tottime:

[image]

Here the overview using snakeviz:

[image]

### I identified the following as possible *easy gains*:

- [ ] SimTelEventSource._fill_trigger_info takes a lot of time for what it does. Reason is the adding of astropy.time.Times for each telescope in each event. (10 % of the full runtime)

- [ ] Instantiating new containers. It is not the creation of containers itself, but the `deepcopy` of the default values. We should avoid creating default containers and then replacing the values, but should directly give the correct attributes.
  For the stage1 tool, also instantiating the `ImageContainer` with nan values only once instead of in each event that is not passing the selection might help (All containers everywhere also make 10%, with deepcopy making over 90% of those 10 %, so again, not creating e.g. a `HillasParameters` container is the problem, but deepcopying complex defaults.)

- [ ] Using `astropy.coordinates.Angle` where a `astropy.units.Quantity` with units rad or deg would be sufficient 

- [ ] Numba compilation, takes a lot of time. Unfortunately for some reason, ctapipe numba code does not seem to be cached for some reason. We should investigate why (`NUMBA_CACHE_DEBUG` / `NUMBA_CACHE_DIR`):
   https://numba.pydata.org/numba-doc/latest/developer/caching.html

- [ ] `CameraGeometry.__getitem__`. I think we should move back to just providing `x` / `y` values, instead of masking the geometry and then taking x/y from the masked geometry. Maybe there is another way to make this faster.

### Not so easy

- [ ] Hillas parameters, most time spent in `astropy.units.Qantity.__new__`, non-negligible time also in `np.cov` and `np.average`. 
       @kosack already started reimplementing this in numba code here: #1512

- [ ] Most time is still used in the `pyeventio`. Since we pretty heavily optimized this using cython and the functions that still take the most time are already implemented in cython, I don't really see easy gains here. But I can try to have a look at it again. 
But this is also in line with the findings of the LAPP team from some years ago, parsing `simtel` is slow, correct me if I'm wrong @vuillaut? 

   2/3 of the runtime are spent in decoding the variable length encoded waveforms. 

- [ ] Extractor runtime, also already pretty heavily optimized.

vuillaut, 2020-11-20:
> But this is also in line with the findings of the LAPP team from some years ago, parsing simtel is slow, correct me if I'm wrong @vuillaut?

Yes, but the solution is not so hard, move to another format :D

maxnoe, 2020-11-20:
Might be hard to do for simtel, but even directly transforming after the simulation would be good, since we read each simtel file probably more than once.

closed 2023-01-04 by maxnoe