cta-observatory/ctapipe #1545: indexing bug in DL1EventSource

kosack, 2020-11-20:
The code here

https://github.com/cta-observatory/ctapipe/blob/a5ad0f181a2f1565324ce6af5c18a806c9b33dd5/ctapipe/io/dl1eventsource.py#L281-L283

adds 1 to the tel index to get the tel_id, but this is not correct in general, particularly if the subarray is a sub-selection of a larger array.   E.g. if we subselect only telescopes `[5,7,9,100]`, the tel_id to index map would be `[5: 0, 1: 7, 2: 9, 3:100]`, which is *not* `tel_index+1`

Therefore this should be changed to use the functions provided by `SubarrayDescription` to take care of this mapping, namely `sub.tel_ids_to_indices(tel_ids)` or `sub.tel_ids_to_mask(tel_ids)`

Example:
```py3
>>> sub = source.subarray.select_subarray("NewSubarray", [1,5,7,10])
>>> sub.tel_ids_to_mask([1,5,7])

array([ True,  True,  True, False])

>>> sub.tel_ids_to_indices([1,5,7,10])

array([0, 1, 2, 3])
```

maxnoe, 2020-11-20:
Indeed. Also this really should be a method in `SubarrayDescription`

kosack, 2020-11-20:
One problem is that `tels_with_trigger` seems not to honor the subarray selected in EventSource, so perhaps fixing it there would help as well (or is it useful know if telescopes outside the subarray triggered, so not sure)

PR #1546 (2020-12-03): Fix DL1 EventSource
Small changes to tools and bugfixes in order to work with the DL1EventSource

- Fix #1545 adding a new method to `SubarrayDescription`
- Fix event generation in the `DL1EventSource`: Every second event was skipped, because the iterator was advanced twice. That's a huge bug if it didn't work for some random other reason before
- Rename `dl1eventsource.mc_headers` to `dl1eventsource.simulation_config` in line with the changes to the `SimtelEventSource` and container structure changes. To make this behave the same way the `SimtelEventSource` does when using e.g. the `stage-1`  [... cut]
closed 2020-12-03 by kosack