cta-observatory/ctapipe #1556: Frames in subarray are not unique to each telescope, could lead to unexpected behavior

kosack, 2020-11-27:
This is just to remind us that we might have to think a bit more about how we use the `CameraFrames` associated with `CameraDescriptions`, particularly for real data, but also for simulations where the pointing direction is not the same per telescope (divergent or mis-pointed).  This is only a problem *if we decide to re-use those CameraFrame objects to include telescope pointing info* (which we do not do so far) 

Right now we store the frame for each telescope in the CameraGeometry.frame attribute. However, for efficiency sake in `SimTelEventSource`, we only have one CameraGeometry per camera type (which are then linked to `subarray.tel[x].camera.geometry`), which means you cannot have a different frame for each telescope.  

In other words:
```py3
geoms = [c.camera.geometry for c in source.subarray.tel.values()]
print(len(geoms))  # 98 (number of telescopes in the run)
unique_geoms = set(geoms)
print(len(unique_geoms))  # 3 (number of cameras)
```

This means we also only have *n_camera_types* CameraFrame objects in the subarray description, which could lead to some confusion if we update them with telescope pointing directions.  I think in HillasReconstructor now, it is not an issue, but could be if we refactor it further.  

TLDR: the developer might expect that there is one frame per telescope, but in fact there is one per *telescope type*, at least for simulations

maxnoe, 2020-11-27:
A first important step would be that `CameraGeometry.transform` does not mutate inplace but returns a new `CameraGeometry`.

maxnoe, 2020-12-01:
Actually, this is what is already done. So if you do:

```
cam_geom_telframe = cam_geom.transform_to(TelescopeFrame(pointing_direction=pointing.tel[tel_id])
```

You get an instance per telescope, not telescope type:
```
from ctapipe.io import EventSource
from ctapipe.coordinates import TelescopeFrame
from astropy.coordinates import AltAz, SkyCoord
from ctapipe.utils import get_dataset_path

source = EventSource(get_dataset_path('gamma_divergent_LaPalma_baseline_20Zd_180Az_prod3_test.simtel.gz'))
subarray = source.subarray
event = next(iter(source))

cameras_tel = {
    tel_id: tel.camera.geometry.transform_to(
        TelescopeFrame(telescope_pointing=SkyCoord(
            az=event.pointing.tel[tel_id].azimuth,
            alt=event.pointing.tel[tel_id].altitude,
            frame=AltAz(),
        ))
    )
    for tel_id, tel in subarray.tel.items()
}

print(cameras_tel[1] is cameras_tel[2])
print(subarray.tel[1].camera.geometry is subarray.tel[2].camera.geometry)

```

HealthyPear, 2020-12-01:
Yes, it seems to be correct at least using the current master.

Did you use a particular setup or an older branch @kosack ?

kosack, 2020-12-02:
No this issue as I said does not affect any code we have now, but I foresee it being a problem in the future.  

For example, when we write "Stage 2" to support real data, we need to update the pointing positions for each event and for each telescope. RIght now, those are in the Frame attributes, and if you loop over telescopes and modify each one's frame, you are actually modifying the same frame for each camera type. 

It means we have to think a bit about how to apply telescope-wise changes to pointing - if they are applied in the instrument model, then we have a problem.  If they are separate, and the instrument stays fixed for the whole run, then it is ok.

maxnoe, 2020-12-02:
@kosack no, that's not true. The frames do not store pointing info. Pointing info is in `event.pointing.tel`

kosack, 2020-12-02:
That's not what I mean: when you want to do a transform that includes the pointing like in the reconstruction, you have to add it as a frame attribute before the transform. For example go from CameraFrame to AltAz, you need to include the telescope pointing.   So at some point you take the pointing from the event and put it into a Frame so that you can do the coordinate transform.

kosack, 2020-12-02:
Related to this, I don't think that the Reconstructor should have to handle pointing corrections and frame construction (even though HillasReconstructor does that now). so it may be nice to have a separate Component that properly generates per-telescope Frames with the correct pointing info from the event.  Or at the very least, we put that code in a base class for Reconstructors or something so that it's not done multiple times.

maxnoe, 2020-12-02:
Fully agree on the second point. Reconstructors should just use a pointing coordinate, not having to construct it. This is connected with the monitoring data and how that is filled into events.

We could think about using a `SkyCoordField` in `event.pointing` that stores itself as two columns in the output for example.

kosack, 2020-12-03:
I think the solution to this issue is just to add some better documentation to avoid pitfalls in the instrument module, but it brings up another one which is to treat pointing in a nicer way, so perhaps we should split that to a new issue.

maxnoe, 2021-05-14:
Fixed in #1426 (camera geometry instance per telescope without a shared frame)

closed 2021-07-27 by kosack