cta-observatory/ctapipe #1558: fix telescope description strings

kosack, 2020-12-02:
## Current Situation

The original version (in very old versions of ctapipe) of the telescope naming in ctapipe.instrument was to have three parts:
* optics type (LST, MST, SST)
* optics variant (1M, GCT for cta; or HESSI, HESSII for HESS, etc)
* camera  name of attached camera

The naming scheme was `<OPTICS>:<CAMERA>` which was broken down into `<TYPE>[-<VARIANT>]:<CAMERA>` (where square brackets show optional parts).  So you would get names like `SST-1M:CHEC`, or `MST:NectarCam`, which were nice descriptive strings.    However, in the end we wanted to use underscores as separators to allow these to be variable names and proper HDF5 dataset names.  

In the *current version*, we get a bit of a strange mix of what was intended and a different scheme.  In the current scheme, there is only optics type (LST, MST, SST) and optics name  (which is either the same as the type, or different in the case of SSTs), and the type is not optional (confusingly). So we get names like

* `LST_LST_LSTCam`
* `MST_MST_NectarCam`
* `MST_MST_FlashCam`
* `SST_ASTRI_CHEC`
* `SST_ASTRI_ASTRICam`

These are generated from attributes in  the subarray description, which can be seen in `subarray.to_table(kind="optics")`:
```
Out[2]:
   description      name type    mirror_area     num_mirrors num_mirror_tiles equivalent_focal_length
                                      m2                                                 m
      str18         str5 str3      float64          int64         int64               float64
------------------ ----- ---- ------------------ ----------- ---------------- -----------------------
SST_ASTRI_ASTRICam ASTRI  SST 14.562566757202148           2                2      2.1500000953674316
    LST_LST_LSTCam   LST  LST  386.7332458496094           1              198                    28.0
  MST_MST_FlashCam   MST  MST 103.83055877685547           1               84                    16.0
```

Further, we may soon change the data model (#1426) to not rely on telescope and camera names, so they should be just for human-readability and for a legend on ArrayDisplays, for example.  In their current form with repeated names, it is not clear what part means what. 

## Some solutions

1. Simplest solution: since the optics names no longer need to be "parseable", perhaps just dropping the "name" field when it is the same as the "type" would be a quick way to get a cleaner naming.  Then you would have `MST_NectarCam`, and `SST_ASTRI_AstriCam` (or we just drop the second part altogether, but then if we look at old files where we do really mix ASTRI and GCT for example, it could be confusing)
1.  if we want to keep parsibility, another option is to just leave blank where there is no  special optics name, e.g. `MST__NectarCam` (with two underscores)
1. Just drop the underscore between the optics type and name and do the same as (2), so you get `MST_NectarCam` and `SSTASTRI_CHEC`
1. change "name" to "version", and then call current MST and LST structures as "v1", and ASTRI as "ASTRI1", wo you get `MST_v1_NectarCam`, `SST_ASTRI1_CHEC`.  That could be nice as we may in the future have upgrades or variants to existing cameras and telescopes.

maxnoe, 2022-09-21:
Just to note that this got worse with Prod6, since now the SST camera is just called `SSTCam` and the SST optics called SST:

```
SST_SST_SSTCam
```

kosack, 2022-10-06:
Since the CTA telescope name and optics structure names seem to always be the same now, perhaps they are no redundant.  What could be more useful is to include an "observatory" (instrument) name + the type of telescope + the camera name. 

E.g: `<ObservatoryName>_<TelescopeType>_<CameraName>`

Then we would get for example:
```
CTAO_SST_SSTCam  
CTAO_MST_FlashCam
CTAO_MST_NectarCam
ASTRI_SST_AstriCam
HESS_LST_NamCam
HESS_MST_
MAGIC_LST_
LST1_LST_LSTCam
```

Tobychev, 2023-01-23:
Do you have an example of where a user has actually complained or of where the current format has lead to a technical issue? 

To me the strings look fine, that they are a bit repetitive is just a accident of the naming scheme or possibly a reason for why the optics should be renamed  ;)

Tobychev, 2023-02-23:
I think this issue is just a dupe of #1140 ? 

In either case, I think the strings are fine and suggest we leave them as is, closing both issues.

maxnoe, 2023-02-23:
Yes, clearly a dupe. On if we should close both, I am not sure, @kosack clearly seems unhappy with the current variant.

I will close this here but leave the other one open for now

closed 2023-02-23 by maxnoe