cta-observatory/ctapipe #1562: DL1Writer should not write pointing info from each event

maxnoe, 2020-12-04:
The DL1Writer should not try to reconstruct the monitoring info by looking into unique pointings. 

Rather, it should have a `write_pointing` method, that gets the info that should be written. 

This requires a common way to store the monitoring info however.

maxnoe, 2020-12-04:
A simple solution would be to implement a `PointingSource` along the lines of `EventSource`, however for simtel files and standard DLx files, this would mean having to repeat the input file (once for `EventSource`, once for `PointingSource`, which is not nice UX.

```python
event_source = EventSource(input_url)
pointing_source = PointingSource(input_url)
writer = DL1Writer(output_file)
for event in source:
    pointing_source(event)
    proces(event)

for pointing in pointing_source:
    writer.write_pointing(pointing)
```

kosack, 2020-12-04:
yeah, this was discussed in a few past Issues as well.  

see #1041  and the diagram here: https://github.com/cta-observatory/ctapipe/issues/1165#issuecomment-548311167

maxnoe, 2020-12-04:
I opened another one because this is partciluar about pointing and the "hack" we currently do in the DL1Writer.

For LST, this has the result of storing pointing info for every event although it would be sufficient to store the drive report values with a resolution of 2s.

kosack, 2020-12-04:
But right now, the DL1 writer does not write per-event pointing, I think, so it should be ok

maxnoe, 2020-12-04:
> But right now, the DL1 writer does not write per-event pointing, I think, so it should be ok.

Because for LST real data the pointing is different for each event, it is written for every event.

kosack, 2020-12-04:
It depends a bit on the design: if we have event-wise (or at least rapidly updated pointing info), we could create a general `MonitoringComponent`  that takes input eventwise containers, and writes out every N events a summary, with columns of value_mean, value_std, value_count to some output stream.

You could add that as a sub-component of DL1Writer, e.g. `DL1Writer.pointing_monitor`, and it would do the right thing

maxnoe, 2020-12-04:
For the lst data, that wouldn't make too much sense since the values come from the drive report with 2 s resolution, so it would make much more sense to just store these data values.

See: https://github.com/cta-observatory/ctapipe_io_lst/blob/ctapipe_compatible/ctapipe_io_lst/pointing.py

kosack, 2020-12-04:
> For the lst data, that wouldn't make too much sense since the values come from the drive report with 2 s resolution, so it would make much more sense to just store these data values.

Yes, even easier then, and no special code needed.   So the question is how to write something at a different rate than the Event stream, when the writing happens in an event loop.  

Options:
* don't write it during event processing, but rather attach the full pointing monitoring info to the file afterward

* add a mechanism to detect when something changes in the pointing, and only call TableWriter.write() if that happens

kosack, 2020-12-04:
I'm a bit confused though: you said:
> For the lst data, that wouldn't make too much sense since the values come from the drive report with 2 s resolution, so it would make much more sense to just store these data values.

but also:
> Because for LST real data the pointing (altaz, tracking ra/dec) is different for each event, it is written for every event.

Which is it?   I would separate pointing corrections from drive reports as well.

maxnoe, 2020-12-04:
> Which is it? I would separate pointing corrections from drive reports as well.

Both actually. It writes the current alt/az and the nominal (not measured) ra/dec.

kosack, 2020-12-04:
I think all of this comes down to mixing a few things.
 
1. it is not the responsibility of the DL1Writer to write hardware level pointing information, that should be written to the DL0 files (can be separate from event data, in another file for example), as well as pointing correction information
2. one of the calibrations done in the DL0-DL1 step should be the application of pointing corrections, and output of "DL1 pointing" information, which is not necessarily the same as the DL0 pointing information generated by the telescope (it can be simplified).  
3. we then need to be able to go from *DL1/Monitoring/Pointing* information to *DL1/Event/Pointing* information, meaning some notion of interpolation of the pointing info back to event time, which has to be more than just a corrected Alt/Az position, it should also include camera rotation and related corrections (focal length, maybe skew if needed for bendy structures).  

So the problem is not just writing back out what the drive reports give, that is DL0 - here we need a step that goes from DL0 → DL1, and puts the pointing in a common place.  No RA/Dec is needed, only Alt/Az, by the way.

kosack, 2020-12-04:
Anyhow, I think this requires a meeting and discussion offline, since it is a general and larger problem and already has several issues related to it.

maxnoe, 2020-12-04:
> it is not the responsibility of the DL1Writer to write hardware level pointing information, that should be written to the DL0 files (can be separate from event data, in another file for example), as well as pointing correction information

Exactly and this is why I opened this issue. We should implement another way to write the pointing info for simtel files so it does not mess with this.

maxnoe, 2024-02-27:
This was fixed in #2438

closed 2024-02-27 by maxnoe