cta-observatory/ctapipe #1563: Storing time does not take scale into account

maxnoe, 2020-12-06:
https://github.com/cta-observatory/ctapipe/blob/f1cd0bb3079bc4d8657bc3917aff09940f8f39ef/ctapipe/io/hdf5tableio.py#L523

It just stores `.mjd` without storing the scale ("utc", "tai", "ut1", ...) that as used.

We should either make sure to store a fixed scale, e.g. by doing `time.tai.mjd` or we should store the scale in the header or both.

kosack, 2020-12-07:
I don't think we should store the scale on an event-wise basis, instead we should add it to the `.meta` field of the container in a consistent way (e.g. `time_scale` and `time_format` keywords)

maxnoe, 2020-12-07:
With "storing a fixed scale" i meant always making sure that the mjd is in e.g. TAI, not storing the scale into a column.

kosack, 2020-12-07:
Yes, that is reasonable.  It could just go in the help string of the Field then, or we use Times as the value in the Field (not sure how slow that is).  Another question is always: is float MJD sufficient for DL1 and above? Or should we store the original high precision timestamp as 2 columns? (or even the float as 2 columns as supported by astropy to reduce numerical rounding errors).  I guess there is no need for such high precision timing after the trigger (at least I can't think of one right now).  Using Times in the field would make it easier to change precision or format later.

kosack, 2020-12-07:
Yes, makes sense - I forgot we already use Times in the field, so just doing the correct conversion is a good idea.

maxnoe, 2020-12-07:
> May also consider if we want a more precise output (2 column values rather than 1).

This would first require making sure that precision isn't lost somewhere along the way.

maxnoe, 2021-02-19:
Fixed in #1586

closed 2021-02-19 by maxnoe