cta-observatory/ctapipe #1564: Switch axes order of TelescopeFrame / NominalFrame for CameraDisplay / CameraGeometry

maxnoe, 2020-12-07:
Today I got an email asking about plotting cameras at the telescope positions and the showers then not matching up with the impact position.

While the email example I got uses `CameraFrame` and not `TelescopeFrame` or `NominalFrame`, this also does not work using those frames, it seems necessary to swap `x / y` for those frames.

Example with a 10 TeV shower on La Palma only showing the 4 LSTs:

```
from ctapipe.visualization import CameraDisplay
from ctapipe.coordinates import TelescopeFrame
from ctapipe.io import EventSource
from ctapipe.calib import CameraCalibrator
import matplotlib.pyplot as plt
import astropy.units as u
import numpy as np

# only the 4 LSTs
allowed_tels = [1, 2, 3, 4]

source = EventSource(
    './Data/sim_telarray/cta-prod5-lapalma/0.0deg/Data/gamma_20deg_180deg_run7___cta-prod5-lapalma_desert-2158m-LaPalma-dark.simtel.zst',
    allowed_tels=allowed_tels,
)

subarray = source.subarray
calib = CameraCalibrator(subarray)

positions = u.Quantity([subarray.positions[tel_id] for tel_id in allowed_tels]).to_value(u.m)

# get relative extent to normalize coordiantes to [0, 1]
min_x, min_y, min_z = positions.min(axis=0)
max_x, max_y, max_z = positions.max(axis=0)

fig = plt.figure(figsize=(10, 10))

ax = fig.add_subplot(1, 1, 1)
ax.scatter(positions[:, 0], positions[:, 1], s=10, zorder=10, label='Telescope Positions')
impact, = ax.plot([], [], 'ro', ms=15, label='impact')
ax.set_aspect(1)
ax.legend()

size = 100
axs = [
    ax.inset_axes([x - size / 2, y - size / 2, size, size], transform=ax.transData)
    for x, y, _ in positions
]

ax.set_xlim(min_x - size / 2, max_x + size / 2)
ax.set_ylim(min_y - size / 2, max_y + size / 2)

disps = []
for i, ax in enumerate(axs):
    geom = subarray.tel[i + 1].camera.geometry

    # transform to telescope frame
    geom_tel = geom.transform_to(TelescopeFrame())

    disps.append(CameraDisplay(geom_tel, ax=ax, cmap='inferno'))

    # remove any labels / axis / ticks
    ax.set_axis_off()
    ax.set_xticks([])
    ax.set_xlabel('')
    ax.set_ylabel('')

fig.tight_layout(pad=0)
fig.show()

for event in source:
    calib(event)
    shower = event.simulation.shower
    impact.set_data(shower.core_x.to_value(u.m), shower.core_y.to_value(u.m))
    for tel_id, dl1 in event.dl1.tel.items():
        disps[tel_id - 1].image = dl1.image

    plt.savefig('shower_image.png')
    break
```

Current master:

[image]

With assigning switching the attribution of `fov_lon / fov_lat` to `x / y`:
[image]

I am not yet sure how I would approach solving this. What I did to produce the image is certainly a hack and not the actual implementation we should use.

maxnoe, 2020-12-07:
Ah, the plot also does not work in `EngineeringCameraFrame`:

[image]

HealthyPear, 2020-12-07:
Why using a frame in the sky to show something on the ground?
As far as I know, the camera frame should be already fixed as the ground frame in corsika, so I think we should limit ourselves to fix that and not the Telescope/Nominal frames

I think this problem could be fixed by checking if (I don't recall now) our ground frame is rotated wrt one from corsika

maxnoe, 2020-12-07:
The `CameraFrame`  is  aligned with Corsika ground frame if and only if the telescope  points to zenith = 0°, az = 0°, and you are right.

This plot should only work in this condition.

HealthyPear, 2020-12-07:
Not sure also why we should support this kind of visualization when we already have e.g.` ArrayDisplay.set_line_hillas`.

This plot is showing essentially the same information (the direction of the image as projected on the ground) plus other information that would be clearer as shown e.g. in the plots used in PR #1408 which would be independent of pointing positions

maxnoe, 2020-12-07:
Because it looks cool.

But you were right, the issue is just the azimuth. My file used 180d and the CameraFrame is only aligned when azimuth is 0d.
You can however rotate the camera by azimuth like so:

```
from ctapipe.visualization import CameraDisplay
from ctapipe.coordinates import TelescopeFrame, EngineeringCameraFrame
from ctapipe.io import EventSource
from ctapipe.calib import CameraCalibrator
import matplotlib.pyplot as plt
import astropy.units as u
import numpy as np
from copy import deepcopy

# only the 4 LSTs
allowed_tels = [1, 2, 3, 4]

source = EventSource(
    './Data/sim_telarray/cta-prod5-lapalma/0.0deg/Data/gamma_20deg_180deg_run7___cta-prod5-lapalma_desert-2158m-LaPalma-dark.simtel.zst',
    allowed_tels=allowed_tels,
)

subarray = source.subarray
calib = CameraCalibrator(subarray)
event = next(iter(source))
calib(event)

positions = u.Quantity([subarray.positions[tel_id] for tel_id in allowed_tels]).to_value(u.m)

# get relative extent to normalize coordiantes to [0, 1]
min_x, min_y, min_z = positions.min(axis=0)
max_x, max_y, max_z = positions.max(axis=0)

fig = plt.figure(figsize=(10, 10))

ax = fig.add_subplot(1, 1, 1)
ax.scatter(positions[:, 0], positions[:, 1], s=10, zorder=10, label='Telescope Positions')
impact, = ax.plot([], [], 'ro', ms=15, label='impact')
ax.set_aspect(1)
ax.legend()

size = 100
axs = [
    ax.inset_axes([x - size / 2, y - size / 2, size, size], transform=ax.transData)
    for x, y, _ in positions
]

ax.set_xlim(min_x - size / 2, max_x + size / 2)
ax.set_ylim(min_y - size / 2, max_y + size / 2)

disps = []
for i, ax in enumerate(axs):

    geom = deepcopy(subarray.tel[i + 1].camera.geometry)
    print(geom.rotate(event.pointing.array_azimuth))

    # transform to telescope frame
    # geom_tel = geom.transform_to(EngineeringCameraFrame())

    disps.append(CameraDisplay(geom, ax=ax, cmap='inferno'))

    # remove any labels / axis / ticks
    ax.set_axis_off()
    ax.set_xticks([])
    ax.set_xlabel('')
    ax.set_ylabel('')

fig.tight_layout(pad=0)
fig.show()

shower = event.simulation.shower
impact.set_data(shower.core_x.to_value(u.m), shower.core_y.to_value(u.m))
for tel_id, dl1 in event.dl1.tel.items():
    disps[tel_id - 1].image = dl1.image

fig.savefig('shower_image_engi.png')
plt.show()
```

So this in fact a non-issue. You can create this kind of plot using ctapipe, there is no bug, you just have to take care of rotation the camera frame to the correct azimuth before plotting.

Closing this.

closed 2020-12-07 by maxnoe

maxnoe, 2020-12-07:
Image using CameraFrame and rotation the cameras by azimuth:
[image]