cta-observatory/ctapipe #1565: Pattern mismatch

Writasree, 2020-12-07:
I'm extracting images using ctapipe from the muonic shower, simulated in CORSIKA at an energy of 1e5 GeV. The coordinates of the four telescopes given in CORSIKA are as follows-
 x(cm)   y(cm)   z(cm)          
 0       -8485     0            TEL. 1
 8485     0        0            TEL. 2
 0        8485     0            TEL. 3
-8485     0        0            TEL. 4
I'm attaching the telescopic images that I've got. I've arranged the telescopic images in accordance with the aforementioned geometric location.

[image]

While I was going through the slides presented by F. Montanet in ESIPAP 2014, I saw the telescopic images (labelled as 1,2,3,4 in green color font) given in slide no 92 ( the screenshot is attached below).

[image]

 Even though the parameters in the input card may differ, the pattern of the image should have matched. I've expected diametrically opposite triggerings for oppositely located telescopes (as I can see in the screenshot). Why does the image obtained via ctapipe doesn't hold the pattern? Am I misunderstanding something or is there any internal rotation happening?

I'm using the following function to extract images-

```
def` display_event(event, source):
      ntels = `len(event.r0.tels_with_data)
      for ii, tel_id in enumerate(event.r0.tels_with_data):
                                nn = int(ceil(sqrt(ntels)))
                                ax = plt.subplot(nn, nn, ii + 1)
                                geom = source.subarray.tel[tel_id].camera.geometry
                                disp = CameraDisplay(geom, ax=ax)
                                disp.pixels.set_antialiaseds(False)
                                disp.autoupdate = False
                                disp.cmap = 'RdBu_r'
                                chan = 0
                                image = event.dl1.tel[tel_id].image
                                mask = tailcuts_clean(source.subarray.tel[tel_id].camera.geometry, image, picture_thresh=10, boundary_thresh=5)
                                cleaned = image.copy()
                                cleaned[~mask] = 0
                                disp.image = cleaned
                                

```
`
`

kosack, 2020-12-07:
It seems your simulations are made with azimuth=180 degrees, which means you also have to rotate your cameras by 180 degrees. Also if the zenith angle is larger than 0, you would have to modify the telescope positions such that they are projected onto a plane normal to the pointing direction.

maxnoe, 2020-12-07:
Please see #1564, where there is an example on how to do this.

closed 2020-12-07 by maxnoe