cta-observatory/ctapipe #1566: Adding a component to Tool.classes breaks giving TelescopeParameter on the command line

maxnoe, 2020-12-09:
It is currently broken to give Telescope parameters on the command line.

The tests I specifically added for this did not catch it because this only happens once the component is added to the `classes` attribute of a `Tool`

maxnoe, 2020-12-09:
```
2020-12-09 16:40:10,346 ERROR [ctapipe.application] (tool.run): Caught unexpected exception: pattern should be a tuple of (command, argument, value)
Traceback (most recent call last):
  File "/home/maxnoe/Uni/CTA/ctapipe/ctapipe/core/tool.py", line 256, in run
    self.setup()
  File "/home/maxnoe/Uni/CTA/ctapipe/ctapipe/core/tests/test_traits.py", line 452, in setup
    self.comp = SomeComponent(subarray=mock_subarray, parent=self)
  File "/home/maxnoe/Uni/CTA/ctapipe/ctapipe/core/component.py", line 237, in __init__
    super().__init__(config, parent, **kwargs)
  File "/home/maxnoe/Uni/CTA/ctapipe/ctapipe/core/component.py", line 141, in __init__
    super().__init__(parent=parent, config=config, **kwargs)
  File "/home/maxnoe/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/traitlets/config/configurable.py", line 104, in __init__
    self.config = config
  File "/home/maxnoe/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/traitlets/traitlets.py", line 604, in __set__
    self.set(obj, value)
  File "/home/maxnoe/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/traitlets/traitlets.py", line 593, in set
    obj._notify_trait(self.name, old_value, new_value)
  File "/home/maxnoe/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/traitlets/traitlets.py", line 1222, in _notify_trait
    type='change',
  File "/home/maxnoe/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/traitlets/traitlets.py", line 1227, in notify_change
    return self._notify_observers(change)
  File "/home/maxnoe/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/traitlets/traitlets.py", line 1264, in _notify_observers
    c(event)
  File "/home/maxnoe/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/traitlets/traitlets.py", line 888, in compatible_observer
    return func(self, change)
  File "/home/maxnoe/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/traitlets/config/configurable.py", line 208, in _config_changed
    self._load_config(change.new, traits=traits, section_names=section_names)
  File "/home/maxnoe/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/traitlets/config/configurable.py", line 175, in _load_config
    setattr(self, name, deepcopy(config_value))
  File "/home/maxnoe/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/traitlets/traitlets.py", line 604, in __set__
    self.set(obj, value)
  File "/home/maxnoe/Uni/CTA/ctapipe/ctapipe/core/traits.py", line 431, in set
    super().set(obj, value)
  File "/home/maxnoe/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/traitlets/traitlets.py", line 578, in set
    new_value = self._validate(obj, value)
  File "/home/maxnoe/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/traitlets/traitlets.py", line 610, in _validate
    value = self.validate(obj, value)
  File "/home/maxnoe/Uni/CTA/ctapipe/ctapipe/core/traits.py", line 393, in validate
    "pattern should be a tuple of (command, argument, value)"
traitlets.traitlets.TraitError: pattern should be a tuple of (command, argument, value)
```

maxnoe, 2020-12-09:
Something makes it that `TelescopeParameter.set` sees a `[value]` list when the component is in `classes`.

kosack, 2020-12-09:
another example of "traitlets.config" is not ideal.

commit f18805f9 (2020-12-09): Add fix for telescope components on cli, fixes #1566
commit 625e16a7 (2021-01-20): Fix 1566, telescope parameter cli (#1567)

* Add failing test for issue 1566

* Add fix for telescope components on cli, fixes #1566

* Move fix to more appropriate position
closed 2021-01-20 by maxnoe