cta-observatory/ctapipe #1575: DL1 format compression methods

maxnoe, 2021-01-14:
Currently, we use float32 for DL1 storage. These values basically do not compress as they are basically random in all the mantissa bits.

However, for the images, even  float32 is more precision than we really need.

We could think about a fixed point representation, that will compress much better (storing scaled integers maybe with an offsett, analog to what is planned for the waveforms).

kosack, 2021-01-14:
If we really want to go for small files, there are also fancier compression methods that do better with data that are mostly sampled noise - see for example Pierre's article on polynomial compression here: https://inspirehep.net/literature/1706075

However, since DL1 is not meant to be archived long term, it might not be worth worrying about it (and the R1/DL0 model is already in fixed-point)

moralejo, 2021-01-14:
@kosack  As you say it is not relevant for the final data flow, since DL1 will not be archived then, but as of now, for LST1 at least, it is very useful to store DL1 files. @maxnoe already did a test in which he saw that the DL1a volume can be ~halved with the simple procedure of using int32 and storing charges and times e.g. in units 0.001 p.e. and 0.001 ns - with some offset if we want to keep the rare negative values (we might as well keep it simple, and just set those to zero at DL1 level). Is there any reason not to implement this? It would not at all affect the standard pipeline, which would never use the DL1 read/write.

kosack, 2021-01-14:
Is there a proper study somewhere of the file sizes vs compression scheme and storage type? Before making a large change to the storage model, it would be good to see some numbers rather than anecdotes.
Also, what is the impact on I/O if values have to be transformed to and from int/float? 

If this is implemented as a *column transform* in the `HDF5TableWriter`, so that the internal representation stays as a float and no API changes are made for the user, then it could be interesting to implement. But that also means that reading the files outside of ctapipe is more difficult, as the user has to apply transformations. For just a scale/shift, it's probably not unreasonable, though. 

If we go this way, I would suggest we allow the option at least to keep negative values, since those are very useful for benchmarking, and we expect to have specialized runs where data-volume-reduction/zero-suppression is turned off that need to be analyzed to study the noise.

kosack, 2021-01-14:
Also keep in mind that eventually we will implement some sort of zero-suppression in DL0, which will carry over to DL1, so it could be that just zeroing pixels that have low signal will give you a similar compression advantage - have you tried that?

moralejo, 2021-01-14:
> Is there a proper study somewhere of the file sizes vs compression scheme and storage type? Before making a large change to the storage model, it would be good to see some numbers rather than anecdotes.

It is clear, for the reason mentioned by Max above, that integers which most often (always, in the case of times) will have low values compared to what fits in an int32, will compress much better than the same values stored as float32. Besides, it has been empirically tested by Max on one real data file. Wouldn't it be extremely surprising that the compression factor depended significantly on the file?

> Also, what is the impact on I/O if values have to be transformed to and from int/float?

Reading the ints and dividing by 1000... does not seem like a big deal. If we implement it we can test it thoroughly. 

> If this is implemented as a _column transform_ in the `HDF5TableWriter`, so that the internal representation stays as a float and no API changes are made for the user, then it could be interesting [...]
For just a scale/shift, it's probably not unreasonable, though.

Indeed. No complication at all (shift is not needed). Instead of labelling the columns 'p.e.' and 'ns' we label them '0.001 pe' and 'ps'.  Easily read outside ctapipe.

> 
> If we go this way, I would suggest we allow the option at least to keep negative values, since those are very useful for benchmarking, and we expect to have specialized runs where data-volume-reduction/zero-suppression is turned off that need to be analyzed to study the noise.

I agree - @maxnoe : we could use signed int32 and then keep also negative values, I guess this will have no impact on compression, right?

kosack, 2021-01-14:
Well, i'ts pretty easy to try it in units if mili-photoelectrons: just create  transform functions:

```py3
def transform_float_to_scaled_int(w):
    return (w*1000.0).astype(np.int32)

def transform_scaled_int_to_float(w):
     return (w/1000.0)
```

and in `DL1Writer` register them with the HDF5TableWriter using:
```py3
   writer.add_column_transform(
        table_name, 
        col_name="image", 
        transform=transform_float_to_scaled_int
   )
```

and of course do the same in DL1EventSource when reading, and perhaps also in `h5_table_to_astropy`

kosack, 2021-01-14:
By the way, storing both waveforms or images differentially is also a way to improve compression, since they are generally mostly noise. That applies more for waveforms of course, but if the pixel ordering is not totally random, it could also help images maybe.  But again, if we finally only store zero-suppressed images, it may give the same benefit.

kosack, 2021-01-14:
sorry closed accidentially

kosack, 2021-01-19:
Related to this, I noticed that for simulations we also store the `true_image` as *float32*, but it is fundamentally an integer quantity in counts so that could be changed to *int32* or even *int16* without any transform needed (that only helps with simulation files of course).

maxnoe, 2021-01-19:
@kosack this is actually a bug in eventio. I initialized the array with the wrong type.

kosack, 2021-01-19:
It's also wrong in `ctapipe.containers` however (it's set to `dtype=np.float32`)

maxnoe, 2021-01-19:
Because that just matches what eventio delivers, otherwise it would raise an error.

maxnoe, 2021-01-19:
@kosack fixed here: https://github.com/cta-observatory/pyeventio/pull/224/

maxnoe, 2021-02-22:
Fixed in #1586

closed 2021-02-22 by maxnoe