cta-observatory/ctapipe #1576: h5_table_to_astropy / read_table loses some column metadata

kosack, 2021-01-19:
# Problem

Since we return an `astropy.table.QTable` instead of `astropy.Table` in `h5_table_to_astropy()` (which is also aliased to `read_table()`), any column that contains a unit loses its description metadata, since those columns are stored as astropy `Quantity` arrays and not `Column` objects. 

```py3
from ctapipe.io import read_table

params = read_table("events.h5", "/dl1/event/telescope/parameters/tel_001")

print(params.columns['obs_id'].description)
print(params.columns['hillas_width'].description)
```
```
observation identifier
AttributeError: 'Quantity' object has no 'description' member
```

Also writing these back out to a file, we see that all the description metadata is lost whenever there is a Quantity column:

```py3
params.write("test.ecsv")
```
```sh
% head test.ecsv

# %ECSV 0.9
# ---
# datatype:
# - {name: obs_id, datatype: int32, description: observation identifier}
# - {name: event_id, datatype: int64, description: event identifier}
# - {name: tel_id, datatype: int16, description: telescope identifier}
# - {name: hillas_intensity, datatype: float64, description: total intensity (size)}
# - {name: hillas_x, unit: m, datatype: float64}
# - {name: hillas_y, unit: m, datatype: float64}
# - {name: hillas_r, unit: m, datatype: float64}
```

A second issue that that QTable is missing some features of Table, like 

# Possible Solution:

Return a  `Table` instead of `QTable` by default.

This makes it slightly less nice to access quantity members with units, since by default the unit is not carried over unless the `quantity` attribute is used:

```
np.nanmean(t['hillas_width'])  # returns a bare float, "0.2"
np.nanmean(t['hillas_width'].quantity) # returns a quantity  "0.2 m"
```

But it does not lose any information.

maxnoe, 2021-01-19:
We should also open an issue about this at astropy.

Maybe a good solution would be that `qtable[key]` always returns a `Quantity` and that `qtable.columns[key]` returns a column.

kosack, 2021-01-26:
This may be fixed already - I updated astropy. and now I can access the description field, even though the return type is still a `Quantity` (seems they now copy the metadata).
```py3
In [16]: params.columns['hillas_width'].description
Out[16]: 'standard spread along the minor-axis'
```
```
params.['hillas_width'].description
```
Also works. However, when writing the table using  `params.write("output.ecsv")` the descriptions are lost.  But I can't reproduce this with my own test table, where when I try exactly the same thing, the column  descriptions are written.  So I need some further investigation before opening an issue with astropy:

This seems to work (where Y is a quantity column, and X is not)
```py3
from astropy.table import Table, QTable, Column
from astropy import units as u

tab = Table({'x':[1, 2, 3], 'y':[5, 6, 7]*u.m})
tab['x'].description = "the X value"
tab['y'].description = "the Y value"
QTable(tab).write("test_qtable.ecsv")
```
```
# %ECSV 0.9
# ---
# datatype:
# - {name: x, datatype: int64, description: the X value}
# - {name: y, unit: m, datatype: float64, description: the Y value}
# schema: astropy-2.0
x y
1 5.0
2 6.0
3 7.0
```

kosack, 2021-01-26:
It seems to be only  problem if the descriptions are assigned *after* converting to a QTable:
```py3
from astropy.table import Table, QTable, Column
from astropy import units as u

tab = QTable({'x':[1, 2, 3], 'y':[5, 6, 7]*u.m})
tab['x'].description = "the X value"
tab['y'].description = "the Y value"
tab.write("test_qtable2.ecsv")
```
Here the description is lost for column Y:
```
# %ECSV 0.9
# ---
# datatype:
# - {name: x, datatype: int64, description: the X value}
# - {name: y, unit: m, datatype: float64}
# meta: !!omap
# - __serialized_columns__:
#     y:
#       __class__: astropy.units.quantity.Quantity
#       unit: !astropy.units.Unit {unit: m}
#       value: !astropy.table.SerializedColumn {name: y}
# schema: astropy-2.0
x y
1 5.0
2 6.0
3 7.0
```

maxnoe, 2021-01-26:
Maybe you were right and it indeed creates more problems than it solves. I would not be opposed to switch back to the plain `Table`.

PR #1605 (2021-02-15): fix read_table
fixes #1576 

read_table returned *QTables* instead of *Tables*, which lead to lots of confusing and annoyances when working with them, since some columns worked differently than others and metadata being lost.  This way all columns in the tables have a consistent interface. 

This just fixes that to return Table (the user can still convert to QTable if needed).  Technically this is an API change, but since the behavior was bug-like, I'm marking it as a bug to be included in v0.10.3. 

Also renamed the function to `read_table` everywhere.  It was already called `ctapipe.io.read_ [... cut]
closed 2021-02-15 by maxnoe