cta-observatory/ctapipe #1577: Add area weight to `SimulatedShower` container

maxnoe, 2021-01-21:
As presented in today's ASWG meeting, the simulations might be performed using impact sampling not uniform per area.

Events using this option will need an additional weight to correctly calculate IRFs (mainly effective area).

The `pyeventio.SimTelFile` delivers this value in  `array_event['mc_event']['aweight']`, this should be propagated into `SimulatedShowerContainer`

kosack, 2021-01-21:
Does this also include the energy spectrum weight? It would be nice to have a single weight for all events.   

This histogram is already there as a second dimension in the thrown energy distribution (written to /simulation/service/shower_distribution in the DL1 files), so in principle it's already possible to correct for this, but having it per event would be much nicer than a histogram per run.

maxnoe, 2021-01-21:
> Does this also include the energy spectrum weight?

No, since the energy spectrum weight depends on the target spectrum, which you can choose freely afterwards.

kosack, 2021-01-21:
I mean correcting the thrown spectrum to a flat one. just like the Aweight corrects to a flat spatial distribution. With both of those weights, you don't need the histograms anymore, but it's not critical (I believe they store such a weight for simulations from ANTARES, etc).  We just have to decide on how we compute the final event weight for IRFs: from summing run header info, from histograms, etc. Right now we assume power-law thrown energy, but that is not generally the case.

maxnoe, 2021-01-21:
Ah right, that would be quite nice I guess.