cta-observatory/ctapipe #1578: No compatible EventSource for merged dl1 data products

Hckjs, 2021-01-25:
meta description is `"Merged DL1 Data Product"` instead of `"DL1 Data Product"`.

kosack, 2021-01-25:
The DESCRIPTION field is meant to be human readable, and shouldn't be relied on to identify the file type.  That should be done checking the other fields like:
* `CTA PRODUCT DATA LEVEL` == "DL1" 
* `CTA PRODUCT DATA MODEL NAME` == "ASWG DL1" 
* `CTA PRODUCT DATA MODEL VERSION` == a compatible version

Hckjs, 2021-01-26:
So checking for
```
if "CTA PRODUCT DATA LEVEL" not in metadata._v_attrnames:
                return False
if metadata["CTA PRODUCT DATA LEVEL"] != "DL1":
                return False
```
in `DL1EventSource.is_compatible()` would be better than
```
if "CTA PRODUCT DESCRIPTION" not in metadata._v_attrnames:
                return False
if metadata["CTA PRODUCT DESCRIPTION"] != "DL1 Data Product":
                return False
```
?

kosack, 2021-01-26:
Yes, those fields have a list of allowed values, whereas the description does not

commit ceb42106 (2021-02-06): Change DL1EventSource.is_compatible() check to meta DATA LEVEL, fixes #1578 (#1579)

* fixes #1578

 - changed meta description

* Change DL1EventSource.is_compatible()

 - Check for DATA LEVEL instead of DESCRIPTION
closed 2021-02-06 by maxnoe