cta-observatory/ctapipe #1594: Remove fits histogram?

maxnoe, 2021-02-10:
* Is unused in any actual code
* Has some tests
* Code looks not really useful

kosack, 2021-02-10:
It's funny - i was just using it to make a look-up table.  It's only feature over just directly using `scipy.stats.binned_statistic2d` or `np.histogramdd` plus a `scipy.interpolate.RegularGridInterpolator` is just that it can write out the resulting histogram to a FITS file, with proper WCS metadata.   Maybe that is not so important.    Oh, and you can also call `fill()` multiple times and it handles that properly: that makes people coming from ROOT happy but is also really useful where you don't want to load an entire dataset at once into memory, since you can make a histogram in chunks (@HealthyPear just ran into that issue with 80gb sample datasets). 

So it could be nice to keep it around, but it would be good to add some functionality to it like the ability to store also the counts values and sum^2 so that all statistics can be computed in one go. 

Nevertheless, we are now implementing the MARS-like direction lookup table in protopipe, so it could still be useful

kosack, 2021-02-10:
We could perhaps replace it with a nice function that writes ND tables with axis info to FITS or HDF files - that is the main functionality needed, not the actual histogram feature.  I guess this is a simliar problem to writing IRF files, so could be a separate library, but there is nothing in scipy/astropy that handles that in a generic way.

maxnoe, 2021-02-10:
I was pointed to it because it emits numpy deprecation warnings.