cta-observatory/ctapipe #1597: Error in stage1 processing

LukasNickel, 2021-02-11:
This has the same root cause as #1411 and #1480 (float32 vs np.nan), but since my symptoms differ, I decided to open a new issue.

When using `ctapipe-stage1` with an `ImageQualityQuery`, that actually removes events (such as in the example configuration), the default container is used for these images, which contains `np.nan` values.
Trying to append such a row to an existing table with data type `float32` fails with the following message:

```python
2021-02-11 13:58:13,425 ERROR [ctapipe.ctapipe-stage1.DL1Writer.HDF5TableWriter] (hdf5tableio._append_row): Error writing col intensity_max of container IntensityStatisticsContainer
2021-02-11 13:58:13,425 ERROR [ctapipe.ctapipe-stage1] (tool.run): Caught unexpected exception: invalid type (<class 'float'>) for column ``intensity_max``
Traceback (most recent call last):
  File "tables/tableextension.pyx", line 1584, in tables.tableextension.Row.__setitem__
TypeError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/lukas/git_repos/ctapipe/ctapipe/core/tool.py", line 260, in run
    self.start()
  File "/home/lukas/git_repos/ctapipe/ctapipe/tools/stage1.py", line 137, in start
    self.write_dl1(event)
  File "/home/lukas/git_repos/ctapipe/ctapipe/io/dl1writer.py", line 209, in __call__
    self._write_telescope_events(self._writer, event)
  File "/home/lukas/git_repos/ctapipe/ctapipe/io/dl1writer.py", line 491, in _write_telescope_events
    *event.simulation.tel[tel_id].true_parameters.values(),
  File "/home/lukas/git_repos/ctapipe/ctapipe/io/hdf5tableio.py", line 317, in write
    self._append_row(table_name, containers)
  File "/home/lukas/git_repos/ctapipe/ctapipe/io/hdf5tableio.py", line 287, in _append_row
    row[colname] = value
  File "tables/tableextension.pyx", line 1589, in tables.tableextension.Row.__setitem__
TypeError: invalid type (<class 'float'>) for column ``intensity_max``
```

No more events are written after this.

This is on python 3.7.9 with pytables 3.6.1

kosack, 2021-02-11:
CAn you say more about how to reproduce this? Is it when running `ctapipe-merge` or when running `ctapipe-stage1`?

LukasNickel, 2021-02-11:
`ctapipe-stage1` with the example config on the file
`gamma_20deg_0deg_run24___cta-prod5b-lapalma_desert-2158m-LaPalma-dark.simtel.zst`
Choosing only telescope id 8 forces this very quickly in my case.

kosack, 2021-02-11:
I guess this only happens when the first event for a telescope is skipped...  I think setting the default to be `np.float32(np.nan)` in containers that expect float32 would fix it.

LukasNickel, 2021-02-11:
Yes, thats what we had in the other issues.
In this case it seems to be the other way around: A number of events is written out just fine and then it tries to append a nan row to the table, that is already setup as `float32` and fails.
The fix is of course the same: We need to make sure, that the default values have the same data type

LukasNickel, 2021-02-11:
Most things are actually `np.float64`:
[image]

See also #1411 

I am not 100% sure why this apparently did not happen earlier. Maybe pytables/numpy implicitly converted `np.nan` to float32 if necessary or there was just noone using the  image quality criteria.

Anyway, its an easy fix. We just need to define which dtypes we want: float64 everywhere would be the easiest for now but probably also a bit of a waste (?)

closed 2021-02-15 by kosack