cta-observatory/ctapipe #1599: flake8 CI

nbiederbeck, 2021-02-12:
After yesterdays developer seminar I linted ctapipe with flake8 as proposed by [github.com/action/starter-workflows](https://github.com/actions/starter-workflows/blob/c59b62dee0eae1f9f368b7011cf05c2fc42cf084/ci/python-package.yml)

```yaml
    - name: Lint with flake8
      run: |
        # stop the build if there are Python syntax errors or undefined names
        flake8 . --count --select=E9,F63,F7,F82 --show-source --statistics
        # exit-zero treats all errors as warnings. The GitHub editor is 127 chars wide
        flake8 . --count --exit-zero --max-complexity=10 --max-line-length=127 --statistics
```

The results are huge (2 and 61 warnings/errors)
```
$ flake8 . --count --select=E9,F63,F7,F82 --show-source --statistics
./ctapipe/utils/unstructured_interpolator.py:96:31: F632 use ==/!= to compare constant literals (str, bytes, int, float, tuple)
        if self._remember and self._previous_v is not 0:
                              ^
./ctapipe/tests/setup_package.py:2:13: F821 undefined name '_ASTROPY_PACKAGE_NAME_'
    return {_ASTROPY_PACKAGE_NAME_ + ".tests": ["coveragerc"]}
            ^
1     F632 use ==/!= to compare constant literals (str, bytes, int, float, tuple)
1     F821 undefined name '_ASTROPY_PACKAGE_NAME_'
2

$ flake8 . --count --exit-zero --max-complexity=10 --max-line-length=127 --statistics
./ctapipe/containers.py:886:60: E241 multiple spaces after ','
./ctapipe/reco/hillas_intersection.py:60:19: E231 missing whitespace after ','
./ctapipe/reco/ImPACT.py:364:5: C901 'ImPACTReconstructor.get_likelihood' is too complex (11)
./ctapipe/reco/ImPACT.py:404:9: F841 local variable 'azimuth' is assigned to but never used
./ctapipe/reco/ImPACT.py:834:13: F841 local variable 'migrad' is assigned to but never used
./ctapipe/reco/tests/test_HillasReconstructor.py:168:50: W291 trailing whitespace
./ctapipe/reco/tests/test_HillasReconstructor.py:170:61: W291 trailing whitespace
./ctapipe/reco/tests/test_HillasReconstructor.py:203:13: F841 local variable 'e' is assigned to but never used
./ctapipe/visualization/mpl_camera.py:87:5: C901 'CameraDisplay.__init__' is too complex (11)
./ctapipe/visualization/mpl_camera.py:219:9: E741 ambiguous variable name 'l'
./ctapipe/visualization/__init__.py:3:43: W291 trailing whitespace
./ctapipe/plotting/bokeh_event_viewer.py:1:1: F401 'numpy as np' imported but unused
./ctapipe/plotting/bokeh_event_viewer.py:3:1: F401 'bokeh.models.Span' imported but unused
./ctapipe/io/hdf5tableio.py:127:5: C901 'HDF5TableWriter._create_hdf5_table_schema' is too complex (13)
./ctapipe/io/hdf5tableio.py:402:5: C901 'HDF5TableReader._map_table_to_containers' is too complex (11)
./ctapipe/io/hdf5tableio.py:447:5: C901 'HDF5TableReader.read' is too complex (14)
./ctapipe/io/dl1eventsource.py:201:5: C901 'DL1EventSource._generate_events' is too complex (23)
./ctapipe/io/tests/test_simteleventsource.py:81:9: F841 local variable 'data' is assigned to but never used
./ctapipe/instrument/subarray.py:7:1: F401 'collections.defaultdict' imported but unused
./ctapipe/instrument/camera/geometry.py:104:5: C901 'CameraGeometry.__init__' is too complex (11)
./ctapipe/utils/download.py:55:9: E722 do not use bare 'except'
./ctapipe/utils/unstructured_interpolator.py:15:1: F401 'time' imported but unused
./ctapipe/utils/unstructured_interpolator.py:96:31: F632 use ==/!= to compare constant literals (str, bytes, int, float, tuple)
./ctapipe/utils/table_interpolator.py:2:86: W291 trailing whitespace
./ctapipe/utils/table_interpolator.py:5:80: W291 trailing whitespace
./ctapipe/utils/table_interpolator.py:9:87: W291 trailing whitespace
./ctapipe/utils/table_interpolator.py:19:80: W291 trailing whitespace
./ctapipe/utils/table_interpolator.py:20:88: W291 trailing whitespace
./ctapipe/utils/table_interpolator.py:79:88: W291 trailing whitespace
./ctapipe/utils/datasets.py:46:1: C901 'find_all_matching_datasets' is too complex (11)
./ctapipe/utils/tests/test_fitshistogram.py:70:28: E231 missing whitespace after ','
./ctapipe/utils/tests/test_CutFlow.py:39:17: E722 do not use bare 'except'
./ctapipe/tools/plot_charge_resolution.py:5:1: F401 'traitlets.Unicode' imported but unused
./ctapipe/tools/display_events_single_tel.py:82:5: C901 'SingleTelEventDisplay.start' is too complex (16)
./ctapipe/tools/dl1_merge.py:176:5: C901 'MergeTool.setup' is too complex (11)
./ctapipe/tools/dl1_merge.py:289:5: C901 'MergeTool.merge_tables' is too complex (14)
./ctapipe/tools/tests/test_tools.py:78:1: C901 'test_merge' is too complex (12)
./ctapipe/tools/tests/__init__.py:4:1: W391 blank line at end of file
./ctapipe/core/traits.py:135:5: C901 'Path.validate' is too complex (15)
./ctapipe/core/traits.py:403:5: C901 'TelescopeParameter.validate' is too complex (12)
./ctapipe/core/tool.py:338:1: C901 'export_tool_config_to_commented_yaml' is too complex (12)
./ctapipe/core/container.py:77:5: C901 'Field.validate' is too complex (13)
./ctapipe/core/provenance.py:54:9: E722 do not use bare 'except'
./ctapipe/core/tests/test_qualityquery.py:33:23: E712 comparison to True should be 'if cond is True:' or 'if cond:'
./ctapipe/image/geometry_converter.py:5:1: F401 '.geometry_converter_hex.convert_geometry_hex1d_to_rect2d' imported but unused
./ctapipe/image/geometry_converter.py:5:1: F401 '.geometry_converter_hex.convert_geometry_rect2d_back_to_hexe1d' imported but unused
./ctapipe/image/geometry_converter.py:10:1: F401 '.geometry_converter_astri.astri_to_2d_array' imported but unused
./ctapipe/image/geometry_converter.py:10:1: F401 '.geometry_converter_astri.array_2d_to_astri' imported but unused
./ctapipe/image/geometry_converter.py:11:1: F401 '.geometry_converter_chec.chec_to_2d_array' imported but unused
./ctapipe/image/geometry_converter.py:11:1: F401 '.geometry_converter_chec.array_2d_to_chec' imported but unused
./ctapipe/image/geometry_converter_hex.py:513:9: E722 do not use bare 'except'
./ctapipe/image/tests/test_cleaning.py:1:1: F401 'os' imported but unused
./ctapipe/image/tests/test_leakage.py:13:5: E741 ambiguous variable name 'l'
./ctapipe/image/tests/test_geometry_converter.py:40:35: E231 missing whitespace after ','
./ctapipe/image/tests/test_geometry_converter.py:84:35: E231 missing whitespace after ','
./ctapipe/tests/setup_package.py:2:13: F821 undefined name '_ASTROPY_PACKAGE_NAME_'
./ctapipe/tests/test_traitles_configurable.py:5:1: F401 'sys' imported but unused
./ctapipe/tests/test_traitles_configurable.py:33:1: C901 'test_all_traitlets_configurable' is too complex (12)
./examples/stereo_reconstruction.py:1:1: F401 'astropy.units as u' imported but unused
./examples/plot_theta_square.py:74:9: E722 do not use bare 'except'
./examples/plot_array_hillas.py:21:1: C901 'If 21' is too complex (13)
18    C901 'Field.validate' is too complex (13)
4     E231 missing whitespace after ','
1     E241 multiple spaces after ','
1     E712 comparison to True should be 'if cond is True:' or 'if cond:'
5     E722 do not use bare 'except'
2     E741 ambiguous variable name 'l'
14    F401 '.geometry_converter_hex.convert_geometry_hex1d_to_rect2d' imported but unused
1     F632 use ==/!= to compare constant literals (str, bytes, int, float, tuple)
1     F821 undefined name '_ASTROPY_PACKAGE_NAME_'
4     F841 local variable 'data' is assigned to but never used
9     W291 trailing whitespace
1     W391 blank line at end of file
61
```

The black pre-commit hook only fixes the single `E241 multiple spaces after ','`.

I don't know how to set this up to only check the proposed changes, but since this is a github workflow maybe it is easier to incoporate than with codacy?

nbiederbeck, 2021-02-12:
While doing this I found out that although we added `pre-commit` to the `cta-dev` environment, we skipped `black` and possible linters.

kosack, 2021-02-15:
That's also taken care of by Codacy (it runs flake8 among about 5-6 other linters)

nbiederbeck, 2021-05-31:
Found a (minor) issue in #1733 that would have been catched by flake8 (or was, but got ignored).

In my opinion, the codacy output is somewhat confusing, and it even seems to get ignored (I could sometimes not even reproduce the errors as shown there).

Should we set up those linters ourselves (e.g. as in top comment) and require them to pass?

nbiederbeck, 2021-05-31:
I think this issue will be addressed and fixed in #1735

closed 2021-05-31 by maxnoe