cta-observatory/ctapipe #1607: stage1 write-index-tables ignores exceptions / ClosedFileError

nbiederbeck, 2021-02-16:
Running `ctapipe-stage1` with `--write-index-tables` results in hundreds of the following block:

```
Exception ignored in: <function Node.__del__ at 0x7f5f50df1550>
Traceback (most recent call last):
  File "/home/nbiederbeck/.local/miniconda3/envs/cta-dl/lib/python3.8/site-packages/tables/node.py", line 319, in __del__
    self._f_close()
  File "/home/nbiederbeck/.local/miniconda3/envs/cta-dl/lib/python3.8/site-packages/tables/table.py", line 2961, in _f_close
    self.flush()
  File "/home/nbiederbeck/.local/miniconda3/envs/cta-dl/lib/python3.8/site-packages/tables/table.py", line 2895, in flush
    self.row._flush_buffered_rows()
  File "tables/tableextension.pyx", line 1337, in tables.tableextension.Row._flush_buffered_rows
  File "tables/tableextension.pyx", line 752, in tables.tableextension.Row.table.__get__
  File "/home/nbiederbeck/.local/miniconda3/envs/cta-dl/lib/python3.8/site-packages/tables/file.py", line 2150, in _check_open
    raise ClosedFileError("the file object is closed")
tables.exceptions.ClosedFileError: the file object is closed
```

kosack, 2021-02-17:
sounds like write_index_tables is being called after the file is closed.  Probably a bug that came up during the refactoring (and I forgot to add a test)

maxnoe, 2025-08-21:
This was fixed.

closed 2025-08-21 by maxnoe