cta-observatory/ctapipe #1610: Store provenance in the actual output file

maxnoe, 2021-02-18:
I think we should not use an extra "provenance log file".

If applicable (e.g. for all tools that produce a single output file where we can store large metadata) we should just store the provenance log, probably in a structured way, in the output file, most importantly for stage 1, we should store the full config.

kosack, 2021-03-05:
Agreed, but it's not always that simple if you output more than one file, have several activities that write to the same file, or if you want to include some fixity data like a checksum that being part of the file changes its own value.  So far, we don't do any of those however, so  I think we could support storing it in the ctapipe DL1/DL2 data model. That doesn't prevent us from also writing it to the log in addition.   Not sure the best way to store a dictionary in HDF5, but just a text blob would probably be ok. 

One use case for the provenance.log was to be able to ingest it into a database in order to track the list of activities that were run. In that case, having it in plain text is nice.   Of course, in its current form, it's a non-standard output file, so there is no real advantage to have it outside the file (it was just for simplicity of implementation), but in general if it would be in  W3C prov format for example, it would be better to have at least an external copy.

kosack, 2021-03-05:
for example, ctapipe-dump-instrument writes many output files.  The provenence log tells you that the *activity* "dump provenance" generates each of these files. 

The *reference metadata* in the header of the DL1 files was meant to be a sort of data product provenance, rather than an activity provenance.

kosack, 2025-01-21:
I'd like to come back to this issue.  We now have a good use case where it should be implemented, and realized it's not as simple as just storing `tool.get_current_config()`, or even just the latest Provenance output in the HDF5 headers or something.  What you really need is _accumulated_ provenance info of some kind, i.e. the full list of provenance from all past activities, perhaps with some de-duplication for merges. 

## Use case:  finding differences in analysis config for existing outputs

After running the full chain of process/merge/train/apply, the user has a file called `electrons_v1.DL2.h5`. Then later, they re-run the same chain with some minor changes to some config parameters, producing `electrons_v2.DL2.h5`.  Many steps were done to make these files.   The user then wants to take the two files and see if there were differences in the configuration _for any step_, i.e. by doing some sort of automatic `diff` between the two provenance trees.  

## Proposed solution

This would of course all be possible with a full _provenance tracker database_ that has ingested all provenance logs, but that is quite a heavy solution and would be a large external system tied to the workflow system more than ctapipe.  However, we could do something more simple:

1. When a tool is run, store the provenance wrapped in a list as a JSON string into the output file.
2. it should be _cumulative_, meaning that the provenance from the activities included in all HDF5 input files should be read and pre-pended to this list, so that the last entry is the current tool's provenance
3. for merges, we could just assume that all inputs have the same info, and just store the first one (this would then drop some history info, like the actual list of `DATA_PRODUCT_IDs` from all inputs, but it would retain at least the changes in configuration)
4. we could filter the provenance info to drop things like `system`, since that section is not so useful here, and takes up a lot of space currently. In fact, even `inputs` may not be needed, due to the comment in step 3. So the main info retained would be the config, the activity name, and the start time, for example. 

The final  data product, like `electrons_v1.DL2.h5`, would then have something like:

```json
{"provenance_history": 
    [
        { "<filtered provenance for activity 1, ctapipe-process>" },
        { "<filtered provenance for activity 2, ctapipe-merge>" },
        { "..." },
        { "<filtered provenance for activity 2, ctapipe-apply-models>" }
    ]
}
```

This way we have the full chain of configuration parameters that were applied, so we even know at the end which image cleaning was used.

maxnoe, 2025-01-21:
> That avoids the var-length-string in table problem for pytables

string attributes have a maximum size though, which is not giant:

```
In [4]: f.root._v_attrs["foo"] = 1000 * "Hello World"

In [5]: f.root._v_attrs["foo"] = 10000 * "Hello World"
---------------------------------------------------------------------------
HDF5ExtError                              Traceback (most recent call last)
Cell In[5], line 1
----> 1 f.root._v_attrs["foo"] = 10000 * "Hello World"

File ~/.local/conda/envs/cta-dev/lib/python3.11/site-packages/tables/attributeset.py:605, in AttributeSet.__setitem__(self, name, value)
    603 def __setitem__(self, name: str, value: Any) -> None:
    604     """Implement a dictionary like interface for `__setattr__()`."""
--> 605     self.__setattr__(name, value)

File ~/.local/conda/envs/cta-dev/lib/python3.11/site-packages/tables/attributeset.py:529, in AttributeSet.__setattr__(self, name, value)
    526     self._g_del_and_log(name)
    528 # Set the attribute.
--> 529 self._g__setattr(name, value)
    531 # Log new attribute addition.
    532 if undo_enabled:

File ~/.local/conda/envs/cta-dev/lib/python3.11/site-packages/tables/attributeset.py:470, in AttributeSet._g__setattr(self, name, value)
    467         stvalue = np.array(value)
    468     value = stvalue[()]
--> 470 self._g_setattr(self._v_node, name, stvalue)
    472 # New attribute or value. Introduce it into the local
    473 # directory
    474 self.__dict__[name] = value

File ~/.local/conda/envs/cta-dev/lib/python3.11/site-packages/tables/hdf5extension.pyx:841, in tables.hdf5extension.AttributeSet._g_setattr()

HDF5ExtError: HDF5 error back trace

  File "H5A.c", line 235, in H5Acreate2
    unable to synchronously create attribute
  File "H5A.c", line 185, in H5A__create_api_common
    unable to create attribute
  File "H5A.c", line 129, in H5A__create_common
    unable to create attribute
  File "H5VLcallback.c", line 1034, in H5VL_attr_create
    attribute create failed
  File "H5VLcallback.c", line 1001, in H5VL__attr_create
    attribute create failed
  File "H5VLnative_attr.c", line 110, in H5VL__native_attr_create
    unable to create attribute
  File "H5Aint.c", line 376, in H5A__create
    unable to create attribute in object header
  File "H5Oattribute.c", line 320, in H5O__attr_create
    unable to create new attribute in header
  File "H5Omessage.c", line 193, in H5O__msg_append_real
    unable to create new message
  File "H5Omessage.c", line 1731, in H5O__msg_alloc
    unable to allocate space for message
  File "H5Oalloc.c", line 1262, in H5O__alloc
    object header message is too large

End of HDF5 error back trace

Can't set attribute 'foo' in node:
 / (RootGroup) np.str_('').

```

kosack, 2025-01-21:
Ok, so needs some storage thought then... 

Another issue: this also means we would need to store the JSON in the model pickle files as well, since those should also be tracked

maxnoe, 2025-01-21:
That's not really an issue, as you can store anything in the pickle, just add it t the instance.

But it might also be a good reason to investigate https://github.com/cta-observatory/ctapipe/issues/1091 again

kosack, 2025-01-21:
Anyhow, this is not super-high priority, but since people will start soon to do some studies with ctapipe, and since this info could also be useful in the testbench reports, it could be interesting to find someone to implement it . For now, we can just manually keep the input config files as part of the testbench input, and of course for any processing done with DIRAC, we know which configs were used.  I think the idea of this issue is more for cases where the processing workflows are not heavily controlled, or where we have many re-processings, so information can be lost if not done automatically.