cta-observatory/ctapipe #1612: Iterating over events in the DL1 source raises error

maxnoe, 2021-02-18:
```
path = './dl1/gamma_20deg_0deg_run100___cta-prod5-paranal_desert-2147m-Paranal-dark_cone10_merged.DL1.h5'
for array_event in EventSource(path):
    print(array_event.index.obs_id, array_event.index.event_id)

```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
tables/tableextension.pyx in tables.tableextension.Row.__getitem__()

KeyError: 'telescopetrigger_time'

During handling of the above exception, another exception occurred:

KeyError                                  Traceback (most recent call last)
tables/tableextension.pyx in tables.tableextension.get_nested_field_cache()

KeyError: 'telescopetrigger_time'

During handling of the above exception, another exception occurred:

KeyError                                  Traceback (most recent call last)
tables/utilsextension.pyx in tables.utilsextension.get_nested_field()

KeyError: 'telescopetrigger_time'

During handling of the above exception, another exception occurred:

KeyError                                  Traceback (most recent call last)
<ipython-input-69-cb09f1e08b84> in <module>
----> 1 for array_event in source:
      2     print(array_event.index.obs_id, array_event.index.event_id)

~/Uni/CTA/ctapipe/ctapipe/io/eventsource.py in __iter__(self)
    272         generator
    273         """
--> 274         for event in self._generator():
    275             yield event
    276             if self.max_events and event.count >= self.max_events - 1:

~/Uni/CTA/ctapipe/ctapipe/io/dl1eventsource.py in _generator(self)
    173 
    174     def _generator(self):
--> 175         yield from self._generate_events()
    176 
    177     def __len__(self):

~/Uni/CTA/ctapipe/ctapipe/io/dl1eventsource.py in _generate_events(self)
    312                     continue
    313                 if self.datamodel_version == "v1.0.0":
--> 314                     data.trigger.tel[i["tel_id"]].time = i["telescopetrigger_time"]
    315                 else:
    316                     data.trigger.tel[i["tel_id"]].time = i["time"]

tables/tableextension.pyx in tables.tableextension.Row.__getitem__()

tables/tableextension.pyx in tables.tableextension.get_nested_field_cache()

tables/utilsextension.pyx in tables.utilsextension.get_nested_field()

KeyError: 'no such column: telescopetrigger_time'

```

LukasNickel, 2021-02-18:
Thats a newly produced file, right?
So this code shouldn't be executed since datamodel_version is > 1.0

LukasNickel, 2021-02-18:
Hm...

maxnoe, 2021-02-18:
THe files have the wrong data model version:
[image]

maxnoe, 2021-02-18:
https://github.com/cta-observatory/ctapipe/blob/bb87bc5019d272241c3392ebe1586268450b8574/ctapipe/tools/dl1_merge.py#L412-L415

maxnoe, 2021-02-18:
After fixing this by hand, I get another error because the table `monitoring.array.pointing` is completely missing the metadata.

LukasNickel, 2021-02-18:
... which is because we forgot to fix this
https://github.com/cta-observatory/ctapipe/blob/bb87bc5019d272241c3392ebe1586268450b8574/ctapipe/tools/dl1_merge.py#L317
in the merge tool. PR is coming

maxnoe, 2021-02-18:
The tool always produced missing metadata, so always wrong.

closed 2021-02-19 by kosack