cta-observatory/ctapipe #1615: Non-updated docs

rlopezcoto, 2021-02-18:
The version displayed in the docs (0.1.dev1+gbb87bc5) is not the current one of ctapipe:
[image]

and also I think that the [Getting Started For Developers](https://cta-observatory.github.io/ctapipe/getting_started/index.html) still recommends a development based on forking the repo, while I understand that, since we cannot use travis anymore, if we want to perform tests using `github ci` we need to do PRs to branches of the repo, instead of the fork. Since people are usually pointed to these instructions when they get started in any cta-observatory package, do you think it would be a good idea to change them?

kosack, 2021-02-18:
Wow, that's an old version of the docs... I guess this is a bug in our new GitHub Actions CI pipeline.

kosack, 2021-02-18:
>  if we want to perform tests using github ci we need to do PRs to branches of the repo, instead of the fork

I'm still using a fork, and it works fine.

HealthyPear, 2021-02-18:
branching or forking shouldn't depend on the CI itself, but rather on operations that require to pass from the official repository, such as downloading protected test data

kosack, 2021-02-18:
I do notice that in the docs builds, there does not seem to be any restictions on when they are deployed - in the previous CI build with Travis, we only deployed docs to GitHub-Pages if this was the master branch, and otherwise the deploy part was skipped.  @maxnoe Is there a way to do that with github actions?  I suspect now they are being overwritten by PR  CI pipelines

HealthyPear, 2021-02-18:
Weird.
I have the same in protopipe and it worked as expected, at least after recent PRs.

maxnoe, 2021-02-18:
@rlopezcoto The reason forks don't work for lstchain or ctapipe_io_lst anymore is the reliance on password protected test data which needs the secrets that are only availlable in the main repo. ctapipe does not use password protected test data.

rlopezcoto, 2021-02-18:
> @rlopezcoto The reason forks don't work for lstchain or ctapipe_io_lst anymore is the reliance on password protected test data which needs the secrets that are only availlable in the main repo. ctapipe does not use password protected test data.
> 
> Another way to solve this would be instead of using secrets would be to use a custom runner for the CI that has the data, which would also work for pull requests from forks. But someone would need to provide a server where that can run.

Ah ok, thanks, that got me confused. In any case, it is a pity that we cannot point people to the "Getting started For Developers" ctapipe link to people that have password protected test data (there may be several, at least for those developing plugins)

maxnoe, 2021-02-18:
@rlopezcoto We should update the docs to state that for regular contributors, we don't recommend forking. At least that is my preferred solution.

rlopezcoto, 2021-02-18:
@maxnoe that sounds good, it would be nice to include this recommendation in the docs if @kosack agrees

maxnoe, 2021-02-19:
>  Maybe we are missing a "git pull --tags" somewhere? 

It's there and they are pulled, I am confused where that version is coming from:

https://github.com/cta-observatory/ctapipe/blob/bb87bc5019d272241c3392ebe1586268450b8574/.github/workflows/ci.yml#L56

> I do notice that in the docs builds, there does not seem to be any restictions on when they are deployed

There is, the deployment is done only on master:

https://github.com/cta-observatory/ctapipe/blob/bb87bc5019d272241c3392ebe1586268450b8574/.github/workflows/ci.yml#L66-L68

You can also see it being skipped here in a PR:

https://github.com/cta-observatory/ctapipe/pull/1614/checks?check_run_id=1929875109

And run on the master here:

https://github.com/cta-observatory/ctapipe/runs/1919537990?check_suite_focus=true

So besides the wrong version, everything works as expected I'd say.

maxnoe, 2021-02-19:
@kosack Interestingly, the git commit hash is correct. bb87bc5 is the current master commit.

HealthyPear, 2021-02-19:
From what I can see there is nothing wrong with the documentation building or deploying, what is wrong is the version number read by the configuration file of the documentation.

I still have to understand how the versioning works now in ctapipe (I wanted to do the same for protopipe, but it appears that it's best for me to see what's the problem  :P).

My point of view:

In `docs/conf.py` we do,

```python
version = ctapipe.__version__
# The full version, including alpha/beta/rc tags.
release = version
```

which is read from `ctapipe/version.py` where, as far as I can see, in this case the version is taken from the section,

```python
try:
    try:
        from ._dev_version import version
    except ImportError:
        from ._version import version
```

Side comment: have you checked for warnings in the compilation of the docs? If yes, perhaps one of these is the one that changes between the `try` and the first `except` (I am excluding the second one since the displayed version is not 0.0.0).
(This is why I prefer to use `readthedocs`, with which I can stop the build if there is **any** warning, even the most stupid)

maxnoe, 2021-02-19:
The version works correctly in a local test.

maxnoe, 2021-02-24:
Version fixed in #1628

closed 2021-02-24 by maxnoe