cta-observatory/ctapipe #1619: DL1Writer cannot handle files containing multiple obs_ids

maxnoe, 2021-02-18:


kosack, 2021-02-19:
I never considered this use case: the assumption was that the DL1Writer (and ctapipe-stage1) would operate only on only single observations, which are then merged, but never to re-read merged DL1 and re-write it out again.   However, I think the same problem will arise when we try to write DL2, since we would want to be able to read a merged DL1 and write a merged DL2 file. 

So I see two options:
1. *Ignore it for now*: the lack of this feature doesn't stop us from writing a pipeline, so we could just defer it to a far future release (and focus on more important features now). In that case, we should raise an exception saying merged files cannot be written. 
2. *Try to support it*: It's not a trivial change (and not just to DL1Writer I think).  The main issue is that all of the header info like the simulation config are single Containers coming from the EventSource, so it would be a large refactoring of EventSource to output multiple configs, etc.  

In option 1, merged files would only be used for the *training*  workflows, i.e. to build models, lookup-tables, and benchmarks.   For the "processing" workflow (DL0 → DL3, with already trained models), we would always operate on single observations for the time being.

maxnoe, 2021-02-19:
@kosack I think it is all implemented but for the writing of the simulation config in the DL1 writer, the DL1 source seems to handle it correctly.

maxnoe, 2021-02-19:
Ah, and the shower distributions

kosack, 2021-02-19:
Ok, sounds like it's less complicated to implement then, and we should just go for it.  Particularly if #1621 is fixed

maxnoe, 2021-02-24:
Fixed in #1614

closed 2021-02-24 by maxnoe