cta-observatory/ctapipe #1624: Subarrays, `allowed_tels` and the Stereo Trigger

maxnoe, 2021-02-19:
This is a an issue concerning the selection of subarrays from the simulated super-arrays.

How do we need to take the stereo trigger into account here? 

E.g. what should happen if we select an event that would have less than 2 telescopes triggered in the set of selected telescopes?

HealthyPear, 2021-02-19:
Are there events with less than 2 triggering telescopes in Prod5?
I never found one in Prod3b.

I would say that until the mono-reconstruction is not in ctapipe, any single-telescope-triggered event (if any) should be recorded and filled with proper image parameters and NaN reconstruction values.

maxnoe, 2021-02-19:
@HealthyPear the problem are not events with less than 2 triggering telescopes in the full array, but there will for sure be events that would never have triggered certain subarrays.

And I was wondering if that is as easy as "check there are at least telescopes in the selected subarray".

HealthyPear, 2021-02-19:
Why do you have to check it before?

This will be catched automatically by the call of any reconstructor, by filling with nan the appropriate containers if the length of the hillas dictionary is <2

so whatever the subarray these events should go through the pipeline anyhow, get calibrated, cleaned and parametrized and not reconstructed if that's the case

am I missing a deeper problem?

maxnoe, 2021-02-19:
Yes, a conceptional problem that an event source should not yield events that would not have triggered the selected subarray.

HealthyPear, 2021-02-19:
so the case you are talking about is the **whole** subarray? not even 1 telescope?

maxnoe, 2021-02-19:
We simulate more telescopes then realistic arrays will have to be able to optimize the layout.

What I want to discuss here is how that decision (simulating more telescopes) affects the stereo trigger and if there is something we need to do about the triggering when selected only a subarray of all the simulated telescopes.

If the answer is "Just check there are at least two triggered telescopes after you selected the subarray", that is simple.

But normally, these trigger decisions are not as simple and we might create biases by using the stereo trigger of the super-array for analysis of a sub-array.

moralejo, 2021-02-19:
> What I want to discuss here is how that decision (simulating more telescopes) affects the stereo trigger and if there is something we need to do about the triggering when selecting only a subarray of all the simulated telescopes.
> 
> If the answer is "Just check there are at least two triggered telescopes after you selected the subarray", that is simple.
> 

This is what we have done in the past when processing CTA MC productions, in which a minimum of "any 2 telescopes" from the super-array was required to write out an event. 
Additionally, for the case of LSTs we had to impose an additional condition: whenever a single LST triggered in the selected subarray, it had to be removed. In this way we simulate the LSTs' hardware stereo trigger. Note that the individual LST's trigger thresholds in the simulation were tuned for the case of hardware stereo trigger (coincidence of any 2 out of 4 LSTs).

maxnoe, 2021-02-19:
@moralejo ok, but that is not an issue anymore for Prod5, right? Since only 4 LSTs are simulated.

moralejo, 2021-02-19:
> @moralejo ok, but that is not an issue anymore for Prod5, right? Since only 4 LSTs are simulated.
> 
> So at least for prod5, requiring at least two telescopes is enough?

No, because in the simtel file you may have events with 1 LST and xxx MSTs. In those events you still have to remove the LST.

HealthyPear, 2021-02-19:
>  in the simtel file you may have events with 1 LST and xxx MSTs. In those events you still have to remove the LST (wouldn't be there in the real data).

Hi @moralejo,
is this true also for your full-array reference analysis? you do not consider any single-LST event?
or is this just for specific subarrays?

maxnoe, 2021-02-19:
> No, because in the simtel file you may have events with 1 LST and xxx MSTs. In those events you still have to remove the LST (wouldn't be there in the real data).

Ok, I didn't realize that this is not implemented at all in the simulation. I thought this problem only arises when removing LSTs that might have contributed.

moralejo, 2021-02-19:
> Ok, I didn't realize that this is not implemented at all in the simulation. I thought this problem only arises when removing LSTs that might have contributed.

Yes, in the simulation it is "any 2 telescopes".

maxnoe, 2021-04-29:
@moralejo coming back to this, so the correct algorithm would be:

1. select subarray
2. Remove events with 1 MST + 1 LST
3. Remove LST telescope events from array events with only 1 LST.

Anything else?

maxnoe, 2021-04-29:
@GernotMaier I am wondering if this might be the reason for the remaining difference of the sensitivity from pyirf and EventDisplay.

How does EventDisplay treat these kinds of Events? And how are they included in the exported DL2 fits files?

GernotMaier, 2021-04-29:
The DL2 contains all events with at least 2 telescopes of each type in a given subarray.

e.g., for a LST + MST array, it could be:
- 2 LSTs, 0 MSTs
- 0 LSTs, 2 MSTs
- 1 LSTs, 2 MSTs
- 2 LSTs, 1MSTs,
- any combination >=2

Events which are in the simulation files, but which do not fulfill any of these criteria are considers as 'no trigger'.

And yes, one could allow 1 LST + 1 MST events, or even 1 LST or 1 MST only. The work on even types and corresponding reconstruction quality should take care of the differences in precision.

maxnoe, 2021-04-29:
My question was if you removed those "invalid" Events for the sensitivity calculation of event display. 1 + 1 would not be a problem, since you require a multiplicity of 4 (now 3?), but do you remove single LST telescope events?

GernotMaier, 2021-04-29:
Yes. It does not fulfill the condition >=2 for at least one telescope type in the subarray.

maxnoe, 2021-04-29:
Sorry, overread the "of each telescope type" part.

kosack, 2021-04-29:
So the cut is just accept if any telescope type has 2 or more telescopes, including cases where there is a single telescope of a single type?

@GernotMaier is there a reason for the 2 of the same type criterion? Is there a problem with just a straight multiplicity cut? I would not think it could cause any problems to have an event with 1 LST + 1 MS;  It should reconstruct fine, as long as they are calibrated properly.

maxnoe, 2021-04-29:
> 1 LST + 1 MS

If I understood correctly, the real stereo trigger requires at least two LSTs for an LST event to be read out. So the real CTA will never have this kind of event. Which means they shouldn't enter sensitivity calculations.

GernotMaier, 2021-04-29:
> So the cut is just accept if any telescope type has 2 or more telescopes, including cases where there is a single telescope of a single type?
>
> @GernotMaier is there a reason for the 2 of the same [...]

No reason (except lazy implementation of reconstruction algorithms) - we should look at e.g., 1 MST + 1 SST events.

maxnoe, 2021-04-30:
> e.g., for a LST + MST array, it could be:
> 1 LSTs, 2 MSTs

But according to @moralejo  the LST should be removed from such events, right?

HealthyPear, 2021-04-30:
> > e.g., for a LST + MST array, it could be:
> > 1 LSTs, 2 MSTs
> 
> But according to @moralejo the LST should be removed from such events, right?

as far as I understood,

- 1LST + 1MST          ----> the entire event is not considered
- 1LST + >=2 MSTs  ----> only the LST image is removed
- 1MST + >=2 LSTs  ----> only the MST image is removed

same for CTAS.

images are remove / not considered even before shower geometry I guess?

maxnoe, 2021-04-30:
> images are remove / not considered even before shower geometry I guess?

I think it should only happen just before combining telescopes to form stereo predictions (either geometric or by combining the ML mono outputs). There is no real benefit of removing them already at the dl1 level and especially for super-array simulation, you want to keep then and then decide at the DL2 level for a specific subarray.

HealthyPear, 2021-04-30:
by "before" shower geometry I meant you record the DL1b data for those to-be-discarder images, but then you don't use them to e.g. estimate the direction or impact parameter

unless I misunderstood what you mean (you fix the images when you use the to estimate the shower geometry which is anyway the first DL2 step)

moralejo, 2021-05-03:
> as far as I understood,
> 
> * 1LST + 1MST          ----> the entire event is not considered
> * 1LST + >=2 MSTs  ----> only the LST image is removed
> * 1MST + >=2 LSTs  ----> only the MST image is removed

No, the last one is wrong. Al three telescopes are kept. There is no hardware stereo for the MST array.

GernotMaier, 2021-05-03:
I am getting more and more confused by this discussions. In the simulations, we have the following trigger conditions:

- at least 2 LSTs
- at least 1 MST

So why do we need to remove events on the analysis level, unless there is software trigger condition foreseen to remove 1 MST events even when there was a valid LST trigger.

moralejo, 2021-05-03:
Hi @GernotMaier, at least at some point the trigger condition in the simulations was "any 2 telescopes", and this required to do this, so that no isolated LSTs were allowed in any event.

GernotMaier, 2021-05-03:
OK - I guess we anyway have to wait for the exact trigger settings. I guess MST trigger rates are anyway on the high site for the prod3b/prod5 simulations and we might expect changes in the trigger threshold after we have experience with the first telescopes.

maxnoe, 2023-02-03:
This is now possible using the `SoftwareTrigger` component introduced in https://github.com/cta-observatory/ctapipe/pull/2136

closed 2023-02-03 by maxnoe