cta-observatory/ctapipe #1625: Horizontal merging of DL1 files

maxnoe, 2021-02-19:
To process observed data, e.g. MAGIC + LST, using stage1, we need a tool to merge DL1 files horizontally (merging subarrays instead of appending events).

kosack, 2021-02-19:
This is actually quite easy, since the data model was designed for this use case.  In fact, it can be done mostly without copying the actual event data if we want, e.g. by creating a new merged file that just has HDF5 data set symlinks to the other files (which would have to be in the same directory).   So having two options for this might be nice: "copy" and  "link".  The first just creates a totally new file with merged data, and the second just creates a symlinked view of existing files, where the various telescopes are in the same place, and only the subarraydescription changes.

jsitarek, 2021-12-22:
I guess by horizontal merging you mean not only putting all the information in one file, but also matching information from the same event from different telescopes. For this please keep in mind that we have different use cases:
1) if we want to match  MAGIC1 and MAGIC2 we should use the stereo event number, that it rather easy
2) if we want to match (one of) MAGIC telescopes with LST1 at least for the time being we need to use time, and this requires some calculated delays. 
Since we want to have all possibilities (M1+LST1, M2+LST1, M1+M2) we would need both behaviours, but probably it is easier to make the code in a way that it does either 1) or 2), and then execute it twice to have all the combinations

kosack, 2022-01-10:
I think what we need is just:
1. a way to combine HDF5 datasets from multiple files into one file (either by copying or linking). That may be even possible using the h5 tools that come with libhdf5 (not sure), but would be a pretty simple tool to write in any case.
2. A tool that is an off-line SWAT that can generate a new `dl1.event.subarray.trigger` table using either the event_id, event time, or other fields to identify coincidences between telescopes in a file.

jsitarek, 2022-01-26:
Hi @kosack 
I thought a bit about this issue with horizontal merging, in particular in the context of MAGIC+LST1, and we also discussed it today among the people involved in LST1+MAGIC analysis. As explained above matching of LST1 to MAGICs is a bit complicated, since you need to use time within a given window and apply configurable delays that can vary between runs and need to be computed first. It is different that matching the MAGIC telescopes together (that is simple because of the common stereo even number) also something that will not be (hopefully) of use to general CTA, so we thought about such a scheme:

in magic-ctapipe we can make the tricky part of assigning the MAGIC stereo trigger number of LST1 events that are matching in time, this number than be added then to the LST1 data. 

In ctapipe horizontal merging can be then implemented easily using a single number (MAGIC stereo trigger number in this case, or some general array trigger number in the future) and we could then merge whatever we want, MAGIC alone, or MAGIC+LST1 or some combination of two telescopes...