cta-observatory/ctapipe #1627: `Container.__repr__` should show values.

maxnoe, 2021-02-19:
The current container `__repr__` is very verbose and especially for containers containing other containers largely unhelpful.

It also doesn't display the actual values, which would be very helpful.

kosack, 2021-02-19:
I originally implemented it that way, but the output became very large in cases where the values were long and so was pretty ugly.  If you use print() you get the values.  I always thought about swapping that though, and having a `Container.ls()` function that prints the contents of the container, and just have repr and str give back the dictionary. 

Might also implement it like `Component._html_repr_()` where you get a nice notebook version.

kosack, 2021-02-22:
One issue I have now is `container.to_dict()`, without recursion enabled, gives you a mix of a dict and the reprs of each class, which is not useful.  Probably  `to_dict()` should *always* recurse, or at least by default (which is what you get with str()).

maxnoe, 2021-05-20:
@kosack I think we should simplify the `__repr__` of a container *instance* to just show the values, essentially making it the same as the dict repr. 

I imagine we should use the current repr for the class, not instances of the class.