cta-observatory/ctapipe #1629: Upload docs of release versions

maxnoe, 2021-02-19:
We should have a version selection to have all documentations of the released versions available

HealthyPear, 2021-02-19:
is it possible to do this from github pages (maybe even for PR?) ?
as far as I know not, and that is why I use readthedocs among other reasons

when we'll move on gitlab afaik there is not even GitLab Pages, so  we would deploy directly on the dataserver but then I have no idea how to make the feature interactive (of course if you dump the docs into a folder then by changing the url to that folder it will work)

maxnoe, 2021-02-19:
> is it possible to do this from github pages (maybe even for PR?) ?
as far as I know not, and that is why I use readthedocs among other reasons

Since github pages is just a place to dump html, in theory it is possible. I don't know if it is easy though.

maxnoe, 2023-01-04:
We enabled readthedocs a while ago: 

https://ctapipe.readthedocs.io

We should probably add some links though...

closed 2023-01-04 by maxnoe