cta-observatory/ctapipe #1632: Support Prod5 subarray selection

kosack, 2021-02-22:
It might be nice to provide some functionality be able to create a subarray from a general SubarrayDescription using the info provided here (for example) for Prod5:

https://gitlab.cta-observatory.org/GernotMaier/CTA-Prod5-PerformanceResults/-/blob/master/Layouts/telescope_lists_south/CTA.prod5S-M1C1-10MSTs30SSTs-MSTF.lis

Something like:

```py3
tel_ids, names = read_subarray_info("CTA.prod5S-M1C1-10MSTs30SSTs-MSTF.lis")
subarray = source.subarray.select_subarray("prod5S-M1C1-10MSTs30SSTs", tel_ids)
```
 
These files also provide human-readable names for the telescopes (e.g. "M-08"), so perhaps adding an attribute for that in the SubarrayDescription could also be useful, and allowing one to name them. Note that that is not the same thing as `str(event.tel[14])`, which returns the name of the TelescopeDescription (e.g. `SST_ASTRI_CHEC`), not the name of the individual telescope

```py3
subarray.set_telescope_names( dict(zip(tel_ids, names)) )
```

That could then be used by the ArrayDisplay to show more useful names than just the tel_id integer.

kosack, 2021-06-03:
this is also related o #1558

maxnoe, 2022-04-19:
With the added support for multiple configuration files, I think we are basically there.

E.g. for a test processing of the Prod5b La Palma data using the Alpha Configuration, I just had this configuration file:

```
# telescope ids of the 4 LST + 9 LST La Palma alpha configuration in Prod5b
# (out of 84 telescopes total)
EventSource:
  allowed_tels: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 19, 35]
```

and called `ctapipe-process` like this:

```
ctapipe-process --config ctapipe_dl2.yml --config prod5b_lapalma_alpha.yml
```

maxnoe, 2022-04-19:
Maybe a way to override the name would be good though:
```
name='MonteCarloArray_1-11,19,35'
```

kosack, 2022-04-19:
Yes, we could add an attribute in EventSource perhaps like `force_subarray_name=Prod5b-alpha` , or even just `subarray_name=` and if it is None, it's either read or  auto-generated from the EventSource, and if not it's set by the user.  That would then propagate to output files nicely.

maxnoe, 2023-01-04:
I think the way this now worked for the data processing is sufficient, e.g. creating config files like these:

```yaml
# telescope ids of the 4 LST + 9 LST La Palma alpha configuration in Prod5b
# (out of 84 telescopes total)
# The layout is called D25 or long form: CTA.prod5N.D25-4LSTs09MSTs-MSTN
# from https://gitlab.cta-observatory.org/GernotMaier/CTA-Prod5-PerformanceResults/-/blob/master/Layouts/telescope_lists_north/CTA.prod5N.D25-4LSTs09MSTs-MSTN.lis
EventSource:
  allowed_tels: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 19, 35]

DataWriter:
  Instrument:
    site: CTA-North
    class_: Subarray
    id_: alpha
    version: Prod5b

```

This avoids having a database of these subarrays inside ctapipe.

Closing for now.

closed 2023-01-04 by maxnoe