cta-observatory/ctapipe #1634: remove charge resolution tools

kosack, 2021-02-23:
We have a few tools like 
* ctapipe-chargeres-extract
* ctapipe-chargeres-hist
* ctapipe-chargeres-plot

Which probably are not needed anymore, as with the DL1 format and benchmarks, you can plot charge resolution pretty easily.   Have to check that ChargeResolutionCalculator does the same as what are in the protopipe benchmarks, but I think it's covered. @HealthyPear and @watsonjj should give the final say though.

HealthyPear, 2021-03-02:
Sorry, I must have overlooked this...

What has been done in protopipe is specific to the comparison with CTAMARS and there are some differences (but I will have to review the code here anyway):

- I calculate an average pixel-wise bias in a safe range of phes
- I calculate the RMS around 1

I guess all these simple operations can be generalized easily and we should compare them to what is already in _cta-benchmarks_/_ctaplot_.

In the next weeks in preparation for the DL1 benchmarking I will review this and start working with @vuillaut to migrate everything I can (also because as soon as I start to refactor _protopipe_'s DL1 part with ctapipe-stage1 I want to start using _cta-benchmarks_ from the GitLab + dataserver new setup)

HealthyPear, 2021-03-02:
Of course, you can remove them already through the PR, I will check out this release to work on the review

commit 977cdea0 (2021-03-08): Merge pull request #1646 from cta-observatory/remove_charge_resolution

Remove charge resolution tools, fixes #1634
closed 2021-03-08 by maxnoe