cta-observatory/ctapipe #1640: definition of peak time returned by ImageExtractors is ambiguous

kosack, 2021-02-28:
The ImageExtractors all return both a charge image and a peak_time image,  however the definition of the peak_time image is ambiguous.  It could have two definitions:
* the peak_time used to position the *integration window* (which is what is currently returned)
* the time of the maximum pulse in the *readout window*

These two are not the same, particularly for method that predict an integration window from the timing slope, e.g. TwoPassWindowSum.  In that case, the peak time that is returned as the time of maximum in the readout window *only* for pixels where the internal image cleaning threshold was exceeded, and for all other pixels you get the predicted peak time

The question is: which peak time should be returned by methods that predict the window from a timing fit?   In ImageProcessor, peak_time is used to get the timing parameters, but then if the image cleaning is not the same as the image cleaning used in the TwoPass method for example, the result is probably not what is expected. 

### Examples:
Peak time returned by LocalPeakWindowSum,  NeighborPeakWindowSum (which you can see is correlated, due to the summing over neighbor waveforms), and TwoPassWindowSum:

[image]

maxnoe, 2021-02-28:
The last one is just wrong. peak time is the time that shall be used for timing parameters. It has to be an extracted peak time for that pixel, not some globally predicted value.

In the extract_around_peak, we implemented it as the weighted sum. The question for algorithms correlating multiple pixels is whether peak time should be calculated on the correlated signal or on the single pixel signal, where I expect there are good arguments for both cases.

kosack, 2021-03-01:
> The last one is just wrong. peak time is the time that shall be used for timing parameters. It has to be an extracted peak time for that pixel, not some globally predicted value.

That's the question here: the return values are not consistent for pixels outside the cleaning mask.  However,  you can use the TwoPass's return value to compute the timing parameters, and you do get the right answer, though it is somewhat redundant as it should just give you the same answer as was derived inside the algorithm (by construction), so this still fits the definition "shall be used for timing parameters", even though it's not at all what I would have expected to get.

kosack, 2021-03-01:
I would suggest in the TwoPass method we just return the peak time image it used internally to compute the time gradient, and then it would solve this. And to update the documentation to state exactly what should be returned.  Now it only says "Floating point pulse time in each pixel in units “ns” Shape: (n_pix)"

HealthyPear, 2021-03-01:
If the output you expect has to be analogous to the first 2 plots you show I would say that we can safely use the 1st pass.
What the TwoPassWindowSum extractor does is anyway keep the (surviving) core pulse times unchanged and modify the other ones in order to see if some neighboring weak charge can be retrieved, but is is *used* only for this.

To be more precise (but maybe we should test this if it makes sense) we could overwrite the pulse time only for pixels in which we effectively retrieved more charge than in 1st pass, while leaving unchanged the other pixels (which are expected to be noise ones anyway).

maxnoe, 2021-03-01:
> should just give you the same answer as was derived inside the algorithm (by construction),

No. The timing features inside the two-pass algorithm are used to redo the extraction. So the charges will change, so the weights will change so the result of the time fit will change.

kosack, 2021-03-01:
the current return value for TwoPass *is* the output of `extract_around_peak`, but extract_around_peak is *called* with the predicted window (which is the correct thing to do for the intensity). 

Yes, I'm ignoring a small change a few percent due to the charges changing, but that is not the point here.

HealthyPear, 2021-03-01:
The final charge returned by TwoPassWindowSum is the one composed of 1st pass core-pixels and (if the image wasn't bright enough from the start) 2nd pass non-core pixels.

Peak time should be of course correlated with the peak, so it should be time-correlated to the extracted charge (weighted sum or whatever - it doesn't matter how since it comes anyway from the same function as the charge).

So we could return:
- 1st-pass pulse time for core pixels surviving the cleaning,
- 2nd-pass pulse time for non-core pixels where charge_2ndPass > charge_1stPass,
- 1st-pass pulse time for non-core pixels where charge_2ndPass <= charge_1stPass.
Whereas in the case of a bright extracted image right from 1st pass we just get that.

maxnoe, 2021-03-01:
@HealthyPear I am very dreadful of several effects when several pixels have their times coming from different methods. 

I think we should provide the same peak time estimation for each pixel and not have three different methods depending on how the two-pass algorithms treated those pixels.

HealthyPear, 2021-03-01:
Well, the *method* of peak-time extraction would be always the same, do you agree? The function is always the same, so every peak-time will have the same exact definition every time.

In the case of 1-pass image extractors, of course, there is no problem, there is just 1 peak time available.

What I propose (to solve the doubts that arose with this discussion) is to use/output the *right* peak time when is the case - aka for 2-pass methods.

This is because the entire point of using 2-pass methods is to *retrieve* more correct information, which in our case is always the pair of floats (charge, time) which by definition need to be correlated.

Thus for me, it is simply **wrong** to use only 1st pass peak-times (in this case) together with a set of charge values of which (possibly) some of them have been corrected via a time-correlation.
By making use of a time-correlation to correct some pixels' charge we are implicitly saying that for those *few* pixels in which we *effectively* retrieve charge the information would have been lost using only 1st pass, so the time for *those* pixels should be the correct(ed) peak time.

HealthyPear, 2021-03-01:
So just to be clearer:

I am in favor to have an output like the first 2 plots that @kosack showed, but when it comes to 2-pass image extraction methods we must remember (and this is my point: *output*) that some pixel information could have been reviewed by the algorithm (correctly if we have retrieved charge from them) and in that case I see no problem in overwriting *only that* information to output.

kosack, 2021-03-01:
The reason I opened this issue is because there are several use cases for the peak time, and it's clear from this conversation that they are still being mixed.   The main cases are:

1. computing the time parameters cleaned image, which is useful for reconstruction (as a second measure of asymmetry, complementary to the 3rd order moments). For that, the return values we have now are probably ok. 
2. for advanced image cleaning (e.g. looking where the peak time is inconsistent, and using that to help determine signal pixels), For that, returning the "recovered" peak time in the TwoPass method is maybe less useful, since you want to see the transition from noise (no clear peak) to clear peak, so the 1st pass peak time would be better (which uses a 3-sample sliding window). 
3.  reconstruction (using e.g. the peak time in an ImPACT fit, etc), where you want to model the image and peak time in clean pixels, and probably a few rings of nearest-neighbor pixels.  For that the current return from TwoPass may be ok, but it's not clear to me. 

One option is to go back to breaking this into two steps:
* TimingImageExtractor, which uses potentially a different algorithm than the ImageIntensityExtractor to derive timing images (peak time, and maybe later width, etc)
* IntensityImageExtractor (which can use the timing image from the previous step, or internally re-compute it (for advanced methods), and returns only intensity

kosack, 2021-03-01:
It might even be interesting for the TimingImageExtractor to return also a measure of the significance of the peak, that way algorithms like the TwoPass extractor can determine which pixels need to have the peak time predicted, and which are stable, without having to resort to a simple intensity threshold as now. 

So the workflow would be:

`TimingImageExtractor` → peak_time, peak_significance → `IntensityImageExtractor` → intensity

maxnoe, 2021-03-01:
I think the main problem here is that the two pass image cleaning is just conceptually doing too much.
It does not simply extract peak sizes and times. It already has an assumption about the signal shape in all pixels. 

So I would say that when using two pass image cleaning, you probably don't want or need to use a cleaning looking at times, since the extraction will already result in low charges where the peak is not at the expected time.

Same for the more advanced templating analyses: you probably don't want to use the two-pass extraction at all, because know you make two, maybe incompatible, model assumptions about the signal.

To conclude: I would say it's fine like it is now. Just we have to keep in mind that the two-pass algorithm already employs a modelling step of the signal that biases the results (maybe such that the reconstruction of small showers gets better).

HealthyPear, 2021-03-01:
It's a bit difficult for me to understand why do we need a different way of defining the peak time.
I think what we have now is sufficient, since we calculate the integrated charge from a window it makes sense that the peak time is just associated to that integration window which is the case now (apart from the special case of extractors that use the absolute highest ADC peak and we just take the window centered there, but in the end is the same concept).

I think having a peak time associated with the final amount of charge should be sufficient to define the output of every extractor. I cannot seem to disentangle the 3 cases you listed...

> It might even be interesting for the TimingImageExtractor to return also a measure of the significance of the peak, that way algorithms like the TwoPass extractor can determine which pixels need to [...]

This is something maybe for another 2-pass image extractor, since that one is bonded to the CTAMARS one.

kosack, 2021-03-01:
I'm ok with sticking with what we have now, but with the caveat that the timing image is not what one expects for things like timing cleaning if you use TwoPass, NeighborPeak (due to correlations), or FixedWindow.   However, I suspect this will lead to confusion in the future, so it's worth thinking about refactoring into two steps, so we can also implement nicer peak-finding algorithms. Right now the filtering is different for each, while in principle it would be nice to use an advanced peak finding filter and still use TwoPass, etc. 

> TimingImageExtractor, which uses potentially a different algorithm than the ImageIntensityExtractor to derive timing images (peak time, and maybe later width, etc), but is consistent in it's output.

If we do go for splitting the methods, this point is moot, since we would break the TwoPass method (and all other methods) into two pieces anyhow - it would be a large refactor.

HealthyPear, 2021-03-01:
why do you say we would break TwoPass?
you are talking about refactoring only `extract_around_peak`, no?

kosack, 2021-03-01:
In my refactoring, twopass would only return intensity, and it's initial timing info would come from the TimingImageExtractor.  

So the "two pass method" would be refactored as two components:  `SlidingWindowTimingImageExtractor(window_size=3`) + `TimeSlopePredictingIntensityImageExtractor`

HealthyPear, 2021-03-01:
OK, still I would like to stress that in case you want to see something like the first 2 plots you showed but using this 2-pass extractor, then the "correct" pulse time final output should be

    1st-pass pulse time for core pixels surviving the cleaning,
    2nd-pass pulse time for non-core pixels where charge_2ndPass > charge_1stPass,
    1st-pass pulse time for non-core pixels where charge_2ndPass <= charge_1stPass.
    Whereas in the case of a bright extracted image right from 1st pass we just get that.

we can't use a 1st pass pulse time with a (corrected) 2nd pass charge