cta-observatory/ctapipe #1642: Make ctapipe work with h5py version 3.0

maxnoe, 2021-03-01:
Right now we restrict to h5py < 3.0 because of how that version changed string writing broke astropy.

We should switch to allowing > 3.0 when astropy supports it.

kosack, 2021-03-01:
Ideally, there would be a  pytables backend for astropy, and we could drop h5py as a dependency. 

We also had mentioned replacing using `astropy.Table.write()` to write the instrument description to HDF5 files and rolling our own solution (which is a bit annoying to have to do, but gives complete control). I would guess in the end that solution could be contributed back to astropy, though.

maxnoe, 2021-03-01:
I discussed this with the astropy maintainers and they were reluctant on adding this kind of non-standardized extension based on hdf5 attributes. 

But using this yaml extension is not really portable...

maxnoe, 2021-03-02:
Relevant PR in astropy, schould be backported to 4.0.5, 4.2.1:
https://github.com/astropy/astropy/pull/11359

PR #1676 (2021-04-15): Require at least astropy 4.0.5, drop restriction to h5py 2.x
Astropy 4.0.5 and 4.2.1 are now released and the meta serialization compatible with pytables again also when using h5py 3.0.

Closes #1642
closed 2021-04-15 by maxnoe