cta-observatory/ctapipe #1645: Use a weighted fit for time parameters?

maxnoe, 2021-03-02:
Coming back from #1639, where a zeroed out image was passed to time parameters.

We don't actually use the image in the time parameters. I assumed we did a weighted fit before going to the robust approach.

Shall we go back to a weighted approach?

If not, we should remove the unused `image` argument

HealthyPear, 2021-03-02:
We could remove `image` from `timing_parameters` and introduce a new sister function where we do first the weighted fit (in that case we should then give those 2 functions some more specific name)

HealthyPear, 2021-03-02:
or we keep `image` and add a keyword boolean argument to apply the weighted fit before `lts_linear_regression`

maxnoe, 2021-03-02:
What do you mean "first do the weighted fit"?

Either you make the fit weighted (and I think we could implement a weighted version of the LTS fit) or not.

HealthyPear, 2021-03-02:
yes nevermind - I vote for having 2 separate functions

kosack, 2021-03-02:
it does make more sense to weight by intensity, so I'd say go for it (could just be an option to the existing function, e.g. `weighted=True`)

HealthyPear, 2021-03-02:
the important thing is that we keep the possibility to choose between both approaches (`TwoPassWindowSum` needs the LTS fit)

so we can either

- do both in one function and switch between them with a keyword argument
- have 2 separate timing functions, one for each algorithm

I vote the second solution

maxnoe, 2021-03-02:
@HealthyPear are you sure MARS does not as well use a weighted LTS approach?

HealthyPear, 2021-03-02:
MARS uses ROOT's "robust" fit option aka LTS regression robust regression which should be the same as implemented in PR #1419

maxnoe, 2021-03-02:
yes, but ROOT can do this weighted: https://root.cern/doc/master/classTGraph.html#aa978c8ee0162e661eae795f6f3a35589

HealthyPear, 2021-03-02:
where do you read that is weighted?
I see only LST without weights in those public members...

plus I think the version of ROOT used by MARS is an "ancient" one, probably 5.34.36
but even [there](https://root.cern/root/html534/src/TGraph.cxx.html#Myd2JD) I could never understand where is the code called by the "ROB" option

maxnoe, 2021-03-02:
It says that if it is a `TGraphErrors` it takes the y-errors into account -> weighting.

But of course I am not sure, I couldn't find the actual code either.

moralejo, 2021-03-02:
> It says that if it is a `TGraphErrors` it takes the y-errors into account -> weighting.
> 
> But of course I am not sure, I couldn't find the actual code either.

It is not weighting by intensity - it is not obvious why that would be necessarily good. IF we happen to know how time resolution depends on intensity, that might be used.

In the CTA analysis in MARS I  just introduced approximate error bars which correspond _more or less_ to the precision of the time measurement, just to have a Chi2 in the right ballpark for the good fits. But the bars are the same for all pixels, so it does not affect the best-fit parameters.

maxnoe, 2021-03-02:
@moralejo thanks for confirming. I would then leave this open as a possible study that could be made, but not an urgent issue.