cta-observatory/ctapipe #1654: `--help-all` does not list Subsubcomponents correctly.

maxnoe, 2021-03-09:
If a `Tool` has nested components, the help does not show the correct options.

E.g. with the LSTEventSource which has a subcomponent `LSTR0Corrections`, the options should be

```
--LSTEventSource.LSTR0Corrections.traitlet
```

But is only shown as

```
--LSTR0Corrections.traitlet
```

kosack, 2021-03-09:
This is due to the fact that you can't have a `classes` attribute for `Configurable` (Component) like there are in Application (Tool).   It would be nice to have that.    In the end, I think we have to make so many modifications to Configurable and Tool that we nearly have our own system - so much for re-using existing tools!

maxnoe, 2021-03-09:
You can, and we use it.

The problem is more that `classes` are a flat list. It would need to be a dict to be able to represent the tree.

kosack, 2021-03-09:
I think we can keep it as a flat list, but just have all Components also have a list of their sub-component classes.  That way it's the responsibility of the Component to declare which configurable classes it uses

Then you traverse the tree by recursively looping through  `tool_or_component.classes`