cta-observatory/ctapipe #1655: HDF5TableWriter cannot append to existing table

maxnoe, 2021-03-10:
Even when opening with using `mode='a'`, the writer cannot actually append to an already existing table.

kosack, 2021-03-11:
I think append only allows one to append new datasets, not to existing tables. What is the use case of this?

kosack, 2021-03-12:
It would have to avoid creating the group and table in append mode.  This was not a use case I supported, to keep the code as simple as possible, but it shouldn't be hard to do if really needed.  I don't quite see why you couldn't just merge existing files, though

maxnoe, 2021-03-12:
It is not a stage1 output. I just made a custom tool to extract some lst counters from multiple files and wanted to append to the same output file.

commit 61865a86 (2021-04-15): Allow appending of rows to existing tables in HDF5TableWriter, fixes #1655
commit 695a0095 (2021-04-15): Merge pull request #1657 from cta-observatory/allow_table_appending

Allow appending of rows to existing tables in HDF5TableWriter, fixes #1655
closed 2021-04-15 by maxnoe