cta-observatory/ctapipe #1660: Make ImageExtractor a TelescopeParameter

maxnoe, 2021-03-19:
I fear that this is not easy / possible with the current configuration system, since it requires both the type and configuration for that `Component`.

But this kind of flexibility is required for solving e.g. #514

watsonjj, 2021-03-19:
Relevant previous discussion:
https://github.com/cta-observatory/ctapipe/issues/702
https://github.com/cta-observatory/ctapipe/issues/703
https://github.com/cta-observatory/ctapipe/issues/1148

maxnoe, 2021-03-19:
Another solution would be to run stage1 multiple times with different `allowed_telescopes` and then horizontally merging that.

kosack, 2021-03-19:
This  is really a step *before* the image extraction: a waveform filtering step, so I think we could implement it that way, and just have two versions: a NullWaveformFilter, and FlashCamWaveformFilter.  Those could just be called by CameraCalibrator.  All you would really need is a BoolTelescopeParameter like "apply_filter"  to start.

maxnoe, 2021-03-19:
@kosack That also sounds like a good solution, however, from the original issue:

> The deconvolved shape is then NOT integrated anymore, but rather evaluated at a certain point in time (typically certainly at it's maximum position). 

So that does not really seem feasible.

kosack, 2021-03-19:
Ah yes. That does make it more difficult.  

The only problem here of course is for Monte-Carlo files, which are already merged by telescope.  For real data, we will process each telescope's DL0 file separately, and then there is no problem having even a totally different configuration for each telescope.      It's not very pretty, but it would be possible to have CameraCalibrator use a TelescopeParameter for extractor_type, and  then in the config, we just assume that you can only have a configuration *per type of extractor* not per telescope type. 

I.e., the config would be (in pseudocode yaml) :
```yaml
CameraCalibrator:
    extractor_type:
        - ["type", "*", "TwoPassWindowSum"]
        - ["type", "*NectarCam", "NectarCamExtractor"]
NectarCamExtractor:
    <parameters...>
TwoPassWindowSum:
   <parameters>
```

maxnoe, 2021-03-19:
@kosack, Ithink that would even work, since the parameters of the different extractors can in turn again be telescope parameters

kosack, 2021-03-19:
yes, you are right, so this seems to be a reasonable way to go.  It just requires modifying CameraCalibrator to construct N image extractors and call the right one.  In principle it does not even need to construct one per tel_id, just one per type.

maxnoe, 2021-03-19:
We need one extractor per extractor class, not telescope types. Since all the telescope specific options are in turn handled again by the specific extractors.

This should be rather easy change, I will open a PR

commit 8258e307 (2021-04-15): Merge pull request #1661 from cta-observatory/extractor_per_telescope

Allow specifying extractors per telescope, fixes #1660
closed 2021-04-15 by maxnoe