cta-observatory/ctapipe #1662: Minor: add easting / northing to array plotting

GernotMaier, 2021-03-22:
Minor suggestion:

add axes titles to the array layout plotting to know where there is North/East:

[image]

maxnoe, 2021-03-22:
x is -> North and y-> West, right?

kosack, 2021-04-15:
Yes, north is on the X-axis in these (simtel convention).  

I think probably we should allow plots  in a more natural frame though, with north upward, to avoid inevitable confusion.  I think the choice of North along X was to make azimuth come out right, so if you use atan2 to compute the angle and quadrant of a point, it is an angle from north, i.e. an astronomical azimuth.

maxnoe, 2023-01-04:
Done in https://github.com/cta-observatory/ctapipe/pull/1911