cta-observatory/ctapipe #1664: Hillas parameters function is not broadcastable

LukasNickel, 2021-03-22:
The `hillas_parameters` function can not be used on an array of images (unlike the num_islands method).
There is actually a check, that enforces a single image.
https://github.com/cta-observatory/ctapipe/blob/83d1a5d5cfc5bb4f79b126152aad3cd0f5cb5b6a/ctapipe/image/hillas.py#L116

I have not tested yet if there is an easy way to add this.

Having this would be very useful for benchmarking.

kosack, 2021-03-22:
A more brute-force option is that I had generated numba version of the code in #1512 ... maybe those could be wrapped using numba.vectorize  or similar so they can be broadcast.  Otherwise, some re-writing may be needed.

maxnoe, 2021-03-22:
The num_islands method is broadcastable?

LukasNickel, 2021-03-22:
Oh, looks like it isn't.
I missed your list comprehension in the workshop notebook.

kosack, 2021-03-23:
it's not really faster than a for-loop, but you can always use map for now (or even parallelized map for very large arrays). You have to make a wrapper since HillasParameters can throw an exception:

```py3
from functools import partial
from  itertools import repeat

def hillas_mappable(geom, im):
    try:
        return hillas_parameters(geom=geom, image=im).as_dict()
    except HillasParameterizationError:
        return HillasParametersContainer().as_dict()
```
then you can do for example:
```py3
table.from_pandas(pd.DataFrame(map(hillas_mappable, repeat(geom), cleaned)))
```
and you get back something that is close, but still not quite right (since each value is a QUantity, rather than having Quantity columns):
[image]

It's not really perfect, since ti

LukasNickel, 2021-03-26:
I have just for fun tried to support broadcasting in the `hillas_parameters`.
It's a bit annoying and the code gets looks slightly harder to read. Its mainly adding axes to the numpy arrays at the appropriate points (which is not always very obvious to the reader of the code).
Unfortunately
```
hillas_parameters(geom[masks], images[masks])
```
doesn't work because you can't index the geometry with a 2D array. However
```
clean = images.copy()
clean[~masks] = 0
hillas_parameters(geom, clean)
```
works.
Speed is identical so unless we plan to make all parameter functions broadcastable in the near future, we can probably close this and keep looping over events.

closed 2021-03-29 by LukasNickel