cta-observatory/ctapipe #1665: Calling Stage1Tool from interactive kernel

vuillaut, 2021-03-23:
The following code
```
from ctapipe.tools.stage1 import Stage1Tool
from traitlets.config import Config

input_file = 'proton_20deg_180deg_run1___cta-prod3-demo-2147m-LaPalma-baseline-mono.simtel.gz'

config = Config(
    {
    'EventSource': {'input_url': input_file,
                   },
        "DL1Writer": {
        "overwrite": True,
        }
    })

tool = Stage1Tool(config=config)

tool.run()
```
called from a script works perfectly but not called from an interactive shell (IPython, Jupyter...)
It seems to produce the file but exits with  an exception `SystemExit: 0`.
Any idea why?

kosack, 2021-03-23:
Yes, the way it currently works means that it throws a SystemExit exception that you have to catch and return as the exit code.  You can just use `ctapipe.core.run_tool` to run it in a script.

maxnoe, 2021-03-23:
This is what we have the `run_tool` function for. It catches the systemexit and returns the value.

```
from ctapipe.tools import run_tool

run_tool(tool)
```

vuillaut, 2021-03-23:
Awesome, thanks!

closed 2021-03-23 by vuillaut

vuillaut, 2021-03-23:
For the record it's 
```from ctapipe.core import run_tool```