cta-observatory/ctapipe #1668: ctapipe (called through lstchain) searches to download from an non-existing ctapipe-extra version

garciagenrique, 2021-03-24:
Hi,

Using lstchain v0.7.0 at la Palma to run the `r0_to_dl1` stage I found the following error.

```
Traceback (most recent call last):
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1273, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/local/home/lstanalyzer/.cache/ctapipe/cccta-dataserver.in2p3.fr/data/ctapipe-extra/v0.3.3'
```

In the env, it is installed ctapipe-extra v0.3.1, but what puzzles me is the version of ctapipe-extra...

the full error is the following

```
Traceback (most recent call last):
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1273, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/local/home/lstanalyzer/.cache/ctapipe/cccta-dataserver.in2p3.fr/data/ctapipe-extra/v0.3.3'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1273, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/local/home/lstanalyzer/.cache/ctapipe/cccta-dataserver.in2p3.fr/data/ctapipe-extra'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1273, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/local/home/lstanalyzer/.cache/ctapipe/cccta-dataserver.in2p3.fr/data'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1273, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/local/home/lstanalyzer/.cache/ctapipe/cccta-dataserver.in2p3.fr'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1273, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/local/home/lstanalyzer/.cache/ctapipe'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1273, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/local/home/lstanalyzer/.cache'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1273, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/local/home/lstanalyzer'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1273, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/local/home'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/bin/lstchain_mc_r0_to_dl1", line 33, in <module>
    sys.exit(load_entry_point('lstchain', 'console_scripts', 'lstchain_mc_r0_to_dl1')())
  File "/fefs/aswg/software/virtual_env/ctasoft/cta-lstchain/lstchain/scripts/lstchain_mc_r0_to_dl1.py", line 69, in main
    custom_config=config,
  File "/fefs/aswg/software/virtual_env/ctasoft/cta-lstchain/lstchain/reco/r0_to_dl1.py", line 570, in r0_to_dl1
    focal = OpticsDescription.from_name(tel_name.split('_')[0]).equivalent_focal_length
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/site-packages/ctapipe/instrument/optics.py", line 98, in from_name
    table = get_table_dataset(optics_table, role="OpticsDescription.from_name")
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/site-packages/ctapipe/utils/datasets.py", line 232, in get_table_dataset
    return try_filetypes(table_name, role, table_types, **kwargs)
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/site-packages/ctapipe/utils/datasets.py", line 178, in try_filetypes
    cache_path = get_cache_path(os.path.join(DEFAULT_URL, filename))
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/site-packages/ctapipe/utils/download.py", line 75, in get_cache_path
    path.parent.mkdir(parents=True, exist_ok=True)
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1277, in mkdir
    self.parent.mkdir(parents=True, exist_ok=True)
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1277, in mkdir
    self.parent.mkdir(parents=True, exist_ok=True)
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1277, in mkdir
    self.parent.mkdir(parents=True, exist_ok=True)
  [Previous line repeated 5 more times]
  File "/fefs/aswg/software/virtual_env/anaconda3/envs/cta/lib/python3.7/pathlib.py", line 1273, in mkdir
    self._accessor.mkdir(self, mode)
PermissionError: [Errno 13] Permission denied: '/local'
```

kosack, 2021-03-24:
ctapipe-extra is no longer used in ctapipe-0.10.x - instead ctapipe download data to a cache directory from the ctapipe-extra directory on the data server, so no ctapipe-extra package should be installed.   But this sounds like a problem with the caching mechanism.  Try removing ~/.local/cache/ctapipe just to be sure, and try again (and also uninstall ctapipe-extra)

maxnoe, 2021-03-24:
lstchain uses `get_dataset_path` at several places also important for actually running analysis, these should just be removed.

kosack, 2021-03-24:
Agreed that you should not use this in lstchain.  You can in principle still use `get_dataset_path` (even though it is really intended for getting test files and default configurations), as long as you set CTAPIPE_SVC_PATH, which can contain any number of directories in which to search - if that is set, it looks first there, before trying to download from the server.   But the intention of the datasets module is access to testing datasets, not real data. 

If we need some real-data search/management functionality, we could open an issue to add that, but that is probably out of scope for ctapipe, as it really depends on the computing infrastructure on which you are runnning. The script that calls ctapipe should handle data management and just pass a valid pathname to the tool.

maxnoe, 2021-03-31:
@kosack should we switch to downloading from the `ctapipe-testdata` directory instead of a directory called `ctapipe-extra` ?

That would make it clearer that this is not related to ctapipe-extra anymore.

I will also go ahead and archive ctapipe-extra

maxnoe, 2021-03-31:
Closing this.

closed 2021-03-31 by maxnoe