cta-observatory/ctapipe #1674: Add geodetic coordinates of array origin to `SubarrayDescription`

maxnoe, 2021-04-07:
For coordinate transforms, we need the geographical location of the arrays. We should add this as an `astropy.coordinates.EarthLocation` attribute to the `SubarrayDescription` and store it in the hdf5 attributes / metadata when serializing the array layout.

For simulations, this could be a traitlet for the simtel event source or left as `None` as long as this information is not available from the files.

kosack, 2021-04-07:
This is definitely needed.  

The problem right now is that we don't even know which site we have when reading the simtel file, which is why it was not done earlier.  And so far the simtel files are not dependent on any earth location, so none is stored in the file. So this is somewhat similar to the issue about having the subarray and site name somehow as metadata.

kosack, 2021-04-07:
This would also allow us to make the `SubarrayDescription.peek` (or perhaps ArrayDisplay in general) make plots like this, which right now took me quite some reverse-engineering to get right:
[image]

maxnoe, 2021-05-05:
Duplicate of #1031

closed 2021-05-05 by maxnoe