cta-observatory/ctapipe #1678: Pixels in FlashCam camera geometry have wrong rotation

BooleanAuthority, 2021-04-09:
After getting the camera geometry of FlashCam via name (`camgeom = CameraGeometry.from_name(camera_name="FlashCam")`)
Pixels are rotated 90deg from the orientation they should be in. I need to correct it manually with `camgeom.pix_rotation= 90 * u.deg`

kosack, 2021-04-09:
That is actually due to the fact that the camgeom.fits files were generated them from prod3 simulations, which were simulated with the incorrect pixel rotation for FlashCam (so it's a simulation bug, not a ctapipe bug).  

However, we should update ctapipe-extra on the data server to use prod5 for these. Until then you can generate your own by taking a simulation file from prod4 and running ctapipe-dump-instrument (which will write out camgeom files for all cameras in the file), and then add the directory where you dumped them to your CTAPIPE_SVC_PATH environment variable. Then ctapipe will look there first, and use the ones you generated.

kosack, 2021-04-09:
Also, you didn't mention why you were using `from_name()` to get the `CameraGeometry`, but in general that should only be used for tests - if you have a simulation file in simtel or DL1 format (or an observation file that can be read by a ctapipe_io plugin), the `CameraGeometry` is part of the `SubarrayDescription` that you get back from the `EventSource`, and that one is constructed directly from the file, not from the test dataset directory. Generally that is what you want to use, not `from_name()`

e.g:
```py3
from ctapipe.io import EventSource
from ctapipe.visualization import CameraDisplay
source = EventSource(some_data_file_readable_by_ctapipe)

flashcam = source.subarray.tel[6].camera
disp = CameraDisplay(flashcam.geometry())
```

[image]

BooleanAuthority, 2021-04-09:
Thank you all for the feedback  👍 
I used `from_name()` since I had no simtelfile. I am currently ray_tracing with simtel and wanted to illustrate the pixels that were hit.  Also the pixel mapping is incorrect when loading `from_name()`. 
For shower_simulations I also get the geometry from the file.

maxnoe, 2021-04-09:
>  I am currently ray_tracing with simtel and wanted to illustrate the pixels that were hit

When you use simtel, why don't you have an output file?

BooleanAuthority, 2021-04-09:
> > I am currently ray_tracing with simtel and wanted to illustrate the pixels that were hit
> 
> When you use simtel, why don't you have an output file?

if one wants to output the photon paths then a textfile is created instead of the usual simtelfile which contains the photon paths.

maxnoe, 2021-04-09:
Then maybe the best solution is to run a simple simulation (e.g. a single calibration event) to get a simtel file that contains the actually used instrument configurations.

kosack, 2021-04-09:
Yes, @maxnoe 's suggestion will be safer, since you are not guaranteed that the pixel ordering in the reference files you get from from_name() is the same as for the simulation you ran.

BooleanAuthority, 2021-04-09:
> Then maybe the best solution is to run a simple simulation (e.g. a single calibration event) with the same configuration to get a simtel file that contains the actually used instrument configurations.

Yep I will do that. Thank you all for the quick replies :)

closed 2021-04-09 by maxnoe