cta-observatory/ctapipe #1683: Missing pixels and extractors (GlobalPeak / NeighborPeak)

maxnoe, 2021-04-22:
LST has now the first observed data with a switched off module.

Initially, we filled the missing pixels with nan, however that broke the GlobalPeak and NeighborPeak extractors since they do a `waveform.mean(axis=-2)` which is then all nan which results in `peak_index = 0` for alle pixels, since `np.argmax([np.nan, np.nan]) == 0`.

I am not proposing to change those algorithms to be able to treat nans, since using `nanmean` is a major increase in runtime.

However, we should probably define in the data model, what value waveforms of `event.mon.tel[tel_id].pixel_status.hardware_failing_pixels` are expected to have and how these are treated by the extractors combining multiple pixels for the extraction.

We probably need to give the extractors this mask and fill zeros for all failing pixels and only look at the non-failing pixels.

kosack, 2021-04-22:
Yes, we should  treat excluded pixels in a consistent way at all data levels.  Also, I guess using hardware_failing_pixels is not enough: that helps with missing drawers, but for extractors that use neighbors/means those would also be affected by noisy pixels as well (stars etc).   Is the current sequence of steps used right now to analyze/calibrate LST data written somewhere? I would like to see where excluded pixels are computed, since that may have some impact on the data model.  E.g. is a pedestal file generated first,  and the pedestal variances used to detect turned-off or noisy pixels?

maxnoe, 2021-04-22:
There are three "failing pixel" arrays, hardware, pedestal and flatfield. The latter two are filled by lstchain by the class that calculates the pedestal and flatfield calibration data on the interleaved pedestal / flat field events.

I don't think this information is yet used for anything. With the exemption of the Muon analysis, lstchain currently uses `LocalPeakWindowSum`, probably exactly for those reasons.

For the cleaning, the levels are adjusted now by the `pedestal_std` from the interleaved events in a second step. 

So first extract images and interleaved calibration info, then calculate image parameters in a second pass.

maxnoe, 2023-01-27:
https://github.com/cta-observatory/ctapipe/pull/2023