cta-observatory/ctapipe #1706: Component to add noise to images

maxnoe, 2021-05-07:
For adapting generic simulations to observation conditions, it might be necessary to add noise to images during the dl1 stage.

This was just implemented in lstchain here https://github.com/cta-observatory/cta-lstchain/pull/708/files and seems to greatly improve the match between observed and simulated data.

kosack, 2021-05-10:
This is what is done in Whipple for many years - it's called "noise padding" and is definitely a good idea to do for mono analysis (for stereo it has less effect). Being able to specify a noise value per pixel would be useful (i.e. for a pair of On/OFF observations, it's useful to increase the noise to the highest of the pedestal variances of each of the two runs in the pair)

kosack, 2021-05-10:
Such a component would also be useful for dealing with broken/noisy pixels, as it lets you inject a factor of 0 to simulate turning off a pixel, and a high value to simulate high NSB from a star, etc

HealthyPear, 2021-05-12:
the implementation inside the component could use the already available `ctapipe.image.toymodel.ImageModel`

maxnoe, 2021-05-12:
@HealthyPear I don't think this has much in common, besides a call to `rng.poisson` or `rng.normal`

watsonjj, 2021-05-13:
I'd like to ask a question with regards to this and #1707.

Would these features replace fully representing the noise and PSF in the MC model? Or are these features intended for exploratory comparisons between simulations and data, such that the MC model can then be appropriately improved?

The current wording in this issue could suggest that this is intended more as a permanently used feature in the pipeline to improve the matching, rather than an exploratory feature, and could lead to MC models that aren't fully representative of the real telescopes.

maxnoe, 2021-05-13:
@watsonjj No, at least for LST, this is only to transform the standard simulations to runs that deviate from the norm due to worse mirror alignment or higher NSB noise.

In FACT we tried simulating only the Cherenkov shower signal and add that to observed pedestals, which could be another possible implementation.

But for now, these simple components would just add additional noise / be a smearing convolution. What you do with it, is the users' choice. You could simulate without any noise and add all required noise using this or simulate a perfect mirror and add all PSF (up to a certain degree) witht he psf smearing component, but that's certainly not the intended use case.

moralejo, 2021-05-13:
I imagine that one could have proper simulations with different NSB levels, in relatively coarse steps, and then fine-tune that with a simple approach like this to achieve even better data-MC matching on a GTI-wise basis. This can include adding stars over the simulated diffuse NSB.  One problem of the simple noise padding approach is that for "biased" pulse integrators (peak-search) the effect of additional NSB would be different  in pixels with a large signal and those with just noise. But I think it is not a showstopper.

As for PSF, it is more for exploratory purposes, I hope that eventually it will be stable enough so that no fine tuning is needed!

kosack, 2021-06-04:
> I imagine that one could have proper simulations with different NSB levels, in relatively coarse steps, and then fine-tune that with a simple approach like this to achieve even better data-MC matching on a GTI-wise basis. 

This is what is planned for CTA.  Also for "tailored" simulations, we would use the actual pedestal measurements as input o the simulations. 

But having  he ability to inject noise during analysis is still useful for studies or fine-tuning.

closed 2022-04-11 by maxnoe