cta-observatory/ctapipe #1707: Compoment to worsen the PSF at DL1a level

maxnoe, 2021-05-07:
Same reason as #1706

kosack, 2021-05-10:
In general it may be nice to have several components (or one big one) that can modify events to worsen them for various studies, e.g where we can introduce random noise into the pixel/waveform values, images (including neighbor pixels effects like cross-talk), image parameters, and reconstruction parameters. ]

This would benefit from some simple coordinated design so that all work in a similar way from a user/config standpoint. I guess for each value type, you would want to specify the random distribution that is applied, for example (Gaussian, Poisson, flat, and their parameters).  This will be useful for e.g. making bracketing IRFs as well.

kosack, 2021-05-10:
Another option is to not make these Components, but just make some tools that modify DL1 (and soon DL2) files to introduce artifacts, and then no change to the pixel loop is needed.

maxnoe, 2021-05-10:
Yes, we can have the components and then a tool that modifies an existing DL1 file using them. 

This was about smearing the DL1a images, so directly worsen the optical PSF, which of course could have an impact on the gamma-ray PSF.

closed 2022-04-11 by maxnoe