cta-observatory/ctapipe #1715: ImPACT's use of iMinuit needs updating

kosack, 2021-05-11:
#1714 switched to use iminuit>=2  in the muon module, but I just noticed it also used (optionally) in ImPACT, and the old API is still being used there (see below).  This also implies we have a missing unit test to check the "minuit" fitting option. 

https://github.com/cta-observatory/ctapipe/blob/4573e88b0015cdf2a9cccd4ba28e722c2256fa16/ctapipe/reco/impact.py#L800-L833

maxnoe, 2021-05-11:
We have a general lack of tests for impact, see #901

maxnoe, 2021-05-11:
Same goes for `ctapipe.image.muon.taubin_circle_fit`. I think It's best to either revert #1714 or fix to use the new API asap.

maxnoe, 2021-05-11:
I'll open a PR for the muon fitter

maxnoe, 2021-05-11:
I am confused, the taubin fit has a test and it fails. How did that not turn up in #1714?

maxnoe, 2025-08-25:
I think this was fixed some time ago.

closed 2025-08-25 by maxnoe